steve-community/steve #919: Hello, I have difficulties to get the GetDiagnostic command working. In particular I do not know how to specify the "location" parameter (URI directory). Colud you please provide some examples and some notes about that? I have tried the ws, http, file:, URI types with no success.....

alegthb, 2022-09-22:
#### Checklist

- [ ] I checked other issues already, but found no answer/solution
- [ ] I checked the documentation and wiki, but found no answer/solution
- [ ] I am running the latest version and the issue still occurs
- [ ] I am sure that this issue is about SteVe (and not about the charging station software or something unrelated to SteVe)

#### Specifications
```
SteVe Version     : ...
Operating system  : ...
JDK               : ...
Database          : ...
```

#### Expected Behavior

...

#### Actual Behavior

...

#### Steps to Reproduce the Problem

  1. ...
  2. ...
  3. ...

#### Additional context

...

alegthb, 2022-09-22:
In particular I do not know how to specify the "location" parameter (URI directory). Colud you please provide some examples and some notes about that? I have tried the ws, http, file:, URI types with no success.....

mosea3, 2022-10-25:
How is it supposed to be? The Charger POSTS the logs to the specified location?

alegthb, 2022-10-26:
Hello,
sorry for the delay in my answer ..... I solved the issue. 
When the OCPP server send the GetDiagnostic request it also has to specify the URL of the FTP location where the charger has to upload the diagnostic file.
In my case I use a local ftp site by Filezilla.
In my case the URL must have the following format: ftp://<server_address>/<server_dir>|username|password. I was not setting the username and password correctly: the | character was missing and the access to the FTP was mot allowed.
Now it is OK.
Thanks
Ale

alegthb, 2022-10-26:
So, I can confirm the GetDiagnostic works great on SteVe and I'm happy about that!

closed 2022-10-26 by alegthb