steve-community/steve #926: org.eclipse.jetty.server.Server - Stopped Server

iicstoday, 2022-09-25:
#### Checklist

- [ ] I checked other issues already, but found no answer/solution
- [ ] I checked the documentation and wiki, but found no answer/solution
- [ ] I am running the latest version and the issue still occurs
- [ ] I am sure that this issue is about SteVe (and not about the charging station software or something unrelated to SteVe)

#### Specifications
```
SteVe Version     : ...
Operating system  : ...
JDK               : ...
Database          : ...
```

#### Expected Behavior

...

#### Actual Behavior

...

#### Steps to Reproduce the Problem

  1. ...
  2. ...
  3. ...

#### Additional context

...

iicstoday, 2022-09-25:
we are try to run steve.jar and I changed all version as per compatiblity but same issue is raised.

goekay, 2022-09-27:
You did not provide any context or details. It's hard to understand the problem.

iicstoday, 2022-09-27:
closed

thank you for help

closed 2022-09-27 by goekay