steve-community/steve #927: ws url sort like ws://127.0.0.1:8080/steve/websocket/CentralSystemService/(chargeBoxId) to ws://127.0.0.1:8080/steve

iicstoday, 2022-09-25:
I change code in steve-3.4.9 in steve-3.4.9\src\main\java\de\rwth\idsg\steve\config\WebSocketConfiguration file
registry.addHandler(handshakeHandler.getDummyWebSocketHandler(), "/websocket/CentralSystemService/")
                .setHandshakeHandler(handshakeHandler)
                .setAllowedOrigins("*");
to
 registry.addHandler(handshakeHandler.getDummyWebSocketHandler(), "/steve")
                .setHandshakeHandler(handshakeHandler)
                .setAllowedOrigins("*");
but not change my url path please give me a solution.
i want to show mu url like this : ws://127.0.0.1:8080/steve

goekay, 2022-09-27:
You cannot achieve this by only changing this, because the root context sits at `/steve`.

iicstoday, 2022-09-27:
but change in which file code how to access path describe please in details.

goekay, 2022-09-27:
Github issues are not a platform for customer service. You are trying to modify the default behavior, and I don't know why it bothers you. Sorry, but you are more or less on your own.

closed 2022-09-27 by goekay

iicstoday, 2022-09-27:
sorry I just asked. 

Thank you for help.