steve-community/steve #928: Deleted Chargebox no transactions in the DB

mare1901, 2022-09-27:
#### Checklist

- [X] I checked other issues already and found no answer
- [X] I checked the documentation and found no answer
- [ ] I am running the latest version and the feature i am requesting is not implemented there

#### Describe the problem you are trying to solve
I deleted a Chargebox from the Steve because of test reasons. Now I don't have one transaction of this chargebox in the DB.
Is this normal, that steve deleteds all transaction data of this Chargebox?

Thanks

goekay, 2022-09-27:
Yes, this is normal. There is even a warning somewhere in UI that this will happen.

mare1901, 2022-09-27:
Okey thanks for your fast reply. By the wy your steve is excellent.

best regards

closed 2022-09-27 by mare1901