steve-community/steve #929: Required support on the project

parvesh-shadow, 2022-09-27:
#### Checklist

- [ ] I checked other issues already, but found no answer/solution
- [ ] I checked the documentation and wiki, but found no answer/solution
- [ ] I am running the latest version and the issue still occurs
- [ ] I am sure that this issue is about SteVe (and not about the charging station software or something unrelated to SteVe)

#### Specifications
```
SteVe Version     : ...
Operating system  : ...
JDK               : ...
Database          : ...
```

#### Expected Behavior

...

#### Actual Behavior

...

#### Steps to Reproduce the Problem

  1. ...
  2. ...
  3. ...

#### Additional context
It is a good project. 
I am looking for a developer who have experience on this code. Who has successfully implemented the project?
We are getting few bugs and challenges in customisations. 
Ready to pay for the support?
...

rahulbhat13, 2022-09-30:
Can contact and suggest the support required

parvesh-shadow, 2022-09-30:
@rahulbhat13 if you have previous experience and capable to work mail me pksaini1990@gmail.com

goekay, 2022-10-01:
i consider this issue resolved, and am closing it.

closed 2022-10-01 by goekay