steve-community/steve #930: How to Configure main.properties file to use it for Secure Websocket connection.

PavanGitt, 2022-09-30:


goekay, 2022-09-30:
if you enable https and set up keystore with certificates, you can use `wss://` instead of `ws://`

PavanGitt, 2022-09-30:
One more thing want to confirm is 
keystore.path - Is this the path of certificate ?
keystore.password- What to provide here?

Any other things that need to be changed for secure websocket connection from Steve side.
If any please add.
Thanks in advance.

goekay, 2022-09-30:
please use google for keystore info and tutorials. it's the standard way to deal with certificates in the java world.

> Any other things that need to be changed for secure websocket connection from Steve side.

nothing comes to my mind right now. obviously, the station should have/know the certificates as well for the https handshake to be successful.

closed 2022-09-30 by goekay