steve-community/steve #939: Problem with "java jar" command. MVN package is done and Build is success.But after executing java command facing an exception which is as follows.

PavanGitt, 2022-10-11:
Starting. FAILED!
Please refer to the log file for details
Exception in thread "main" java.util.concurrent.ExecutionException: java.lang.IllegalStateException: STOPPED
    at java.base/java.util.concurrent.CompletableFuture.reportGet(CompletableFuture.java:395)
    at java.base/java.util.concurrent.CompletableFuture.get(CompletableFuture.java:2022)
    at org.eclipse.jetty.server.Server.doStop(Server.java:489)
    at org.eclipse.jetty.util.component.AbstractLifeCycle.stop(AbstractLifeCycle.java:132)
    at de.rwth.idsg.steve.JettyServer.stop(JettyServer.java:180)
    at de.rwth.idsg.steve.SteveProdStarter.stop(SteveProdStarter.java:80)
    at de.rwth.idsg.steve.Application.stop(Application.java:85)
    at de.rwth.idsg.steve.Application.main(Application.java:68)
Caused by: java.lang.IllegalStateException: STOPPED
    at org.eclipse.jetty.server.handler.ContextHandler.shutdown(ContextHandler.java:772)
    at java.base/java.util.stream.ReferencePipeline$3$1.accept(ReferencePipeline.java:195)
    at java.base/java.util.ArrayList$ArrayListSpliterator.forEachRemaining(ArrayList.java:1655)
    at java.base/java.util.stream.AbstractPipeline.copyInto(AbstractPipeline.java:484)
    at java.base/java.util.stream.AbstractPipeline.wrapAndCopyInto(AbstractPipeline.java:474)
    at java.base/java.util.stream.AbstractPipeline.evaluate(AbstractPipeline.java:550)
    at java.base/java.util.stream.AbstractPipeline.evaluateToArrayNode(AbstractPipeline.java:260)
    at java.base/java.util.stream.ReferencePipeline.toArray(ReferencePipeline.java:517)
    at org.eclipse.jetty.util.component.Graceful.shutdown(Graceful.java:146)
    ... 6 more

has context menu

goekay, 2022-10-18:
it is impossible to understand your problem from the post. this exception stack trace is not helpful.

closed 2022-10-19 by PavanGitt

xiaojieTao, 2023-08-28:
Have you fixed the problem? I am having the exact same problem now.

Chloe-ko, 2024-01-31:
> Have you fixed the problem? I am having the exact same problem now.

I've had the same issue just now and it turned out to be because I was trying to run Steve on a privileged port (80) while the user it was running as didn't have the rights to start anything on a privileged port.