steve-community/steve #945: Optionally allow database connection without SSL

steinlec, 2022-10-17:
#### Checklist

- [x] I checked other issues already and found no answer
- [x] I checked the documentation and found no answer
- [x] I am running the latest version and the feature i am requesting is not implemented there

#### Describe the problem you are trying to solve

In my setup, there is no need to enable SSL for accessing the databse. So I would be very pleased to can disable that feature in the main.properties.

#### Describe the solution you'd like

I have made some changes to the code:

   --- a/pom.xml
   +++ b/pom.xml
   @@ -42,7 +42,7 @@
            <jackson.version>2.13.2</jackson.version>
    
            
   -        <jdbcUrl>jdbc:mysql://${db.ip}:${db.port}/${db.schema}?useSSL=true&amp;serverTimezone=UTC</jdbcUrl>
   +        <jdbcUrl>jdbc:mysql://${db.ip}:${db.port}/${db.schema}?useSSL=${db.useSSL}&amp;serverTimezone=UTC</jdbcUrl>
        </properties>
    
        <profiles>
   --- a/src/main/java/de/rwth/idsg/steve/config/BeanConfiguration.java
   +++ b/src/main/java/de/rwth/idsg/steve/config/BeanConfiguration.java
   @@ -95,7 +95,7 @@ public class BeanConfiguration implements WebMvcConfigurer {
            hc.addDataSourceProperty(PropertyKey.prepStmtCacheSqlLimit.getKeyName(), 2048);
            hc.addDataSourceProperty(PropertyKey.characterEncoding.getKeyName(), "utf8");
            hc.addDataSourceProperty(PropertyKey.connectionTimeZone.getKeyName(), CONFIG.getTimeZoneId());
   -        hc.addDataSourceProperty(PropertyKey.useSSL.getKeyName(), true);
   +        hc.addDataSourceProperty(PropertyKey.useSSL.getKeyName(), CONFIG.getUseSSL());
    
            // https://github.com/RWTH-i5-IDSG/steve/issues/736
            hc.setMaxLifetime(580_000);
   --- a/src/main/java/de/rwth/idsg/steve/SteveConfiguration.java
   +++ b/src/main/java/de/rwth/idsg/steve/SteveConfiguration.java
   @@ -83,6 +83,7 @@ public enum SteveConfiguration {
                   .schema(p.getString("db.schema"))
                   .userName(p.getString("db.user"))
                   .password(p.getString("db.password"))
   +               .useSSL(p.getBoolean("db.useSSL"))
                   .sqlLogging(p.getBoolean("db.sql.logging"))
                   .build();
    
   @@ -171,6 +172,7 @@ public enum SteveConfiguration {
            private final String schema;
            private final String userName;
            private final String password;
   +        private final boolean useSSL;
            private final boolean sqlLogging;
        }
    
   --- a/src/main/resources/config/prod/main.properties
   +++ b/src/main/resources/config/prod/main.properties
   @@ -11,6 +11,7 @@ db.port = 3306
    db.schema = stevedb
    db.user = steve
    db.password = changeme
   +db.useSSL = false
    
    # Credentials for Web interface access
    #
   --- a/src/main/resources/config/docker/main.properties
   +++ b/src/main/resources/config/docker/main.properties
   @@ -11,6 +11,7 @@ db.port = 3306
    db.schema = stevedb
    db.user = steve
    db.password = changeme
   +db.useSSL = true
    
    # Credentials for Web interface access
    #

HiGregory, 2022-10-17:
@steinlec SSL is common for the production use of OCPP.

Are you suggesting an optional DB connection for sandbox or testing purposes?

steinlec, 2022-10-18:
@HiGregory Of course for production use. I need it for testing purpose on localhost. So I do not need SSL in the database setup.