steve-community/steve #946: Update "Installation" to add autostart within Ubuntu Container

Jeepersfinest, 2022-10-18:
Hi,
i use steve on Ubuntu in a container (Qnap) environemnt. Works fine.
is it possible to explain how to run steve on start automaticlly?

Thanks in advanced
Christian

shadsnz, 2022-10-18:
We run SteVe using an Ubuntu VM and start it automatically with the following cron task, with the simple schedule set to run on boot:

java -jar /home/admin/steve-3.4.5/target/steve.jar >> /home/admin/logs/steve-cron.log &

Jeepersfinest, 2022-11-05:
any further ideas? I´m looking forward to any usefull tips!

Anmirazik, 2023-03-16:
Hi , can you please share your docker main.properties file ? I am not able to run using docker successfully , it shows error as below 

> 2023-03-16 22:35:35 [INFO ] 2023-03-16 14:35:35,691 de.rwth.idsg.steve.utils.PropertiesFileLoader (main) - Hint: The Java system property 'main.properties' can be set to point to an external properties file, which will be prioritized over the bundled one
2023-03-16 22:35:35 [INFO ] 2023-03-16 14:35:35,764 de.rwth.idsg.steve.Application (main) - Loaded the properties. Starting with the 'PROD' profile
2023-03-16 22:35:35 [INFO ] 2023-03-16 14:35:35,778 de.rwth.idsg.steve.Application (main) - Date/time zone of the application is set to UTC. Current date/time: 2023-03-16T14:35:35.768Z
2023-03-16 22:35:35 Log file: Not available
2023-03-16 22:35:35 Starting.[INFO ] 2023-03-16 14:35:35,987 org.eclipse.jetty.server.Server (main) - jetty-10.0.13; built: 2022-12-07T20:13:20.134Z; git: 1c2636ea05c0ca8de1ffd6ca7f3a98ac084c766d; jvm 11.0.4+11
2023-03-16 22:35:36 [INFO ] 2023-03-16 14:35:36,165 org.eclipse.jetty.server.handler.ContextHandler.steve (main) - No Spring WebApplicationInitializer types detected on classpath
2023-03-16 22:35:36 [INFO ] 2023-03-16 14:35:36,195 org.eclipse.jetty.server.session.DefaultSessionIdManager (main) - Session workerName=node0
2023-03-16 22:35:36 [INFO ] 2023-03-16 14:35:36,203 org.eclipse.jetty.server.handler.ContextHandler.steve (main) - Initializing Spring root WebApplicationContext
2023-03-16 22:35:36 [INFO ] 2023-03-16 14:35:36,203 org.springframework.web.context.ContextLoader (main) - Root WebApplicationContext: initialization started
2023-03-16 22:35:36 .[INFO ] 2023-03-16 14:35:36,878 com.zaxxer.hikari.HikariDataSource (main) - HikariPool-1 - Starting...
2023-03-16 22:35:36 [INFO ] 2023-03-16 14:35:36,997 com.zaxxer.hikari.pool.HikariPool (main) - HikariPool-1 - Added connection com.mysql.cj.jdbc.ConnectionImpl@2296127
2023-03-16 22:35:36 [INFO ] 2023-03-16 14:35:36,999 com.zaxxer.hikari.HikariDataSource (main) - HikariPool-1 - Start completed.
2023-03-16 22:35:38 ..[INFO ] 2023-03-16 14:35:38,007 org.apache.cxf.wsdl.service.factory.ReflectionServiceFactoryBean (main) - Creating Service {urn://Ocpp/Cs/2010/08/}CentralSystemService from class ocpp.cs._2010._08.CentralSystemService
2023-03-16 22:35:38 .[INFO ] 2023-03-16 14:35:38,357 org.apache.cxf.endpoint.ServerImpl (main) - Setting the server's publish address to be /CentralSystemServiceOCPP12
2023-03-16 22:35:38 [INFO ] 2023-03-16 14:35:38,366 org.apache.cxf.wsdl.service.factory.ReflectionServiceFactoryBean (main) - Creating Service {urn://Ocpp/Cs/2012/06/}CentralSystemService from class ocpp.cs._2012._06.CentralSystemService
2023-03-16 22:35:38 [INFO ] 2023-03-16 14:35:38,410 org.apache.cxf.endpoint.ServerImpl (main) - Setting the server's publish address to be /CentralSystemServiceOCPP15
2023-03-16 22:35:38 [INFO ] 2023-03-16 14:35:38,411 org.apache.cxf.wsdl.service.factory.ReflectionServiceFactoryBean (main) - Creating Service {urn://Ocpp/Cs/2015/10/}CentralSystemService from class ocpp.cs._2015._10.CentralSystemService
2023-03-16 22:35:38 [INFO ] 2023-03-16 14:35:38,435 org.apache.cxf.endpoint.ServerImpl (main) - Setting the server's publish address to be /CentralSystemServiceOCPP16
2023-03-16 22:35:38 [INFO ] 2023-03-16 14:35:38,436 org.apache.cxf.wsdl.service.factory.ReflectionServiceFactoryBean (main) - Creating Service {urn://Ocpp/Cs/2010/08/}CentralSystemService from class ocpp.cs._2010._08.CentralSystemService
2023-03-16 22:35:38 [INFO ] 2023-03-16 14:35:38,440 org.apache.cxf.endpoint.ServerImpl (main) - Setting the server's publish address to be /CentralSystemService

goekay, 2024-08-07:
@Jeepersfinest recently [this change](https://github.com/steve-community/steve/pull/1508) was introduced. i think this is the solution you are looking for. therefore, i am closing the issue.

@Anmirazik i think your issue is not related. plus, i dont see any issue. contrary to your claim, i dont see error in logs you posted.

closed 2024-08-07 by goekay