steve-community/steve #947: Configuring the Development Environment

cwhzzang, 2022-10-19:
I want to know about Steve's development environment setting. If you have a document on the setting of the development environment, please let me know. Thank you.

goekay, 2022-10-19:
there is a dev profile which you can use with the [respective configs](https://github.com/steve-community/steve/tree/master/src/main/resources/config/dev).

build using the dev profile:
```
# mvn package -Pdev
```

and then run the jar file as usual.

juherr, 2022-10-30:
@cwhzzang You should have a look at #420 because you need to generate classes from jooq if you want to be able to compile and run the project.

goekay, 2025-09-15:
i consider this issue resolved, and am closing it.

closed 2025-09-15 by goekay