steve-community/steve #955: Duplicate entries in transaction_stop

tvand, 2022-10-31:
#### Checklist

- [x] I checked other issues already, but found no answer/solution
- [x] I checked the documentation and wiki, but found no answer/solution
- [ ] I am running the latest version and the issue still occurs
- [ ] I am sure that this issue is about SteVe (and not about the charging station software or something unrelated to SteVe)

#### Specifications
```
SteVe Version     : 3.4.9
Operating system  : Docker Image from amazoncorretto:11.0.15-alpine3.15
JDK               : 11.0.15
Database          : 10.3.32-MariaDB
```

#### Expected Behaviour
One entry per transaction id in transaction_stop

#### Actual Behavior
Duplicate entries with slightly varying event_timestamp

#### Steps to Reproduce the Problem

For some reason, the charging station sends the StartTransaction command for the same transaction id several times:
```
[INFO ] 2022-10-28 20:15:27,169 de.rwth.idsg.steve.ocpp.ws.WebSocketLogger - [chargeBoxId=Wall-E, sessionId=ea7f8986-0e64-bc9a-8052-35ecee81c321] Received: [2,"1243","StatusNotification",{"connectorId":1,"errorCode":"NoError","status":"Preparing"}]                                                                                                                      
[INFO ] 2022-10-28 20:15:27,365 de.rwth.idsg.steve.ocpp.ws.WebSocketLogger - [chargeBoxId=Wall-E, sessionId=ea7f8986-0e64-bc9a-8052-35ecee81c321] Sending: [3,"1243",{}]                                                                            
[INFO ] 2022-10-28 20:15:34,206 de.rwth.idsg.steve.ocpp.ws.WebSocketLogger - [chargeBoxId=Wall-E, sessionId=ea7f8986-0e64-bc9a-8052-35ecee81c321] Received: [2,"1244","StatusNotification",{"connectorId":1,"errorCode":"NoError","status":"Charging"}]                                                                                                                       
[INFO ] 2022-10-28 20:15:34,238 de.rwth.idsg.steve.ocpp.ws.WebSocketLogger - [chargeBoxId=Wall-E, sessionId=16198537-6497-7fdd-79aa-2106f44e33c9] Received: [2,"122004","StartTransaction",{"connectorId":1,"idTag":"00000000000000","meterStart":0,"timestamp":"2022-10-28T20:15:33.561+00:00"}]                                                                             
[INFO ] 2022-10-28 20:15:34,245 de.rwth.idsg.steve.ocpp.ws.WebSocketLogger - [chargeBoxId=Wall-E, sessionId=4c692f63-c4d9-78f2-1257-bafc7555846e] Received: [2,"2777","StartTransaction",{"connectorId":1,"idTag":"00000000000000","meterStart":0,"timestamp":"2022-10-28T20:15:33.561+00:00"}]                                                                               
[INFO ] 2022-10-28 20:15:35,230 de.rwth.idsg.steve.ocpp.ws.WebSocketLogger - [chargeBoxId=Wall-E, sessionId=208f10a1-7595-7d0b-4424-494a9c372333] Received: [2,"37039","StartTransaction",{"connectorId":1,"idTag":"00000000000000","meterStart":0,"timestamp":"2022-10-28T20:15:33.561+00:00"}]                                                                              
[INFO ] 2022-10-28 20:15:37,224 de.rwth.idsg.steve.ocpp.ws.WebSocketLogger - [chargeBoxId=Wall-E, sessionId=ea7f8986-0e64-bc9a-8052-35ecee81c321] Sending: [3,"1244",{}]                                                                            
[INFO ] 2022-10-28 20:15:37,938 de.rwth.idsg.steve.ocpp.ws.WebSocketLogger - [chargeBoxId=Wall-E, sessionId=ea7f8986-0e64-bc9a-8052-35ecee81c321] Received: [2,"1245","StartTransaction",{"connectorId":1,"idTag":"00000000000000","meterStart":0,"timestamp":"2022-10-28T20:15:33.561+00:00"}]                                                                               
[INFO ] 2022-10-28 20:15:39,989 de.rwth.idsg.steve.ocpp.ws.WebSocketLogger - [chargeBoxId=Wall-E, sessionId=4c692f63-c4d9-78f2-1257-bafc7555846e] Sending: [3,"2777",{"transactionId":28,"idTagInfo":{"status":"Accepted","expiryDate":"2022-10-28T21:15:36.452Z","parentIdTag":"default"}}]                                                                                  
[INFO ] 2022-10-28 20:15:39,989 de.rwth.idsg.steve.ocpp.ws.WebSocketLogger - [chargeBoxId=Wall-E, sessionId=208f10a1-7595-7d0b-4424-494a9c372333] Sending: [3,"37039",{"transactionId":28,"idTagInfo":{"status":"Accepted","expiryDate":"2022-10-28T21:15:36.452Z","parentIdTag":"default"}}]                                                                                 
[INFO ] 2022-10-28 20:15:39,990 de.rwth.idsg.steve.ocpp.ws.WebSocketLogger - [chargeBoxId=Wall-E, sessionId=16198537-6497-7fdd-79aa-2106f44e33c9] Sending: [3,"122004",{"transactionId":28,"idTagInfo":{"status":"Accepted","expiryDate":"2022-10-28T21:15:36.452Z","parentIdTag":"default"}}]                                                                                
[INFO ] 2022-10-28 20:15:40,607 de.rwth.idsg.steve.ocpp.ws.WebSocketLogger - [chargeBoxId=Wall-E, sessionId=ea7f8986-0e64-bc9a-8052-35ecee81c321] Sending: [3,"1245",{"transactionId":28,"idTagInfo":{"status":"Accepted","expiryDate":"2022-10-28T21:15:37.944Z","parentIdTag":"default"}}]                                                                                  
```

That works fine and only one transaction is actually started in transaction_start because the transaction id is the primary key in that table.

When the transaction is stopped, however, slightly different event_timestamps cause duplicate entries in transaction_stop:

```
[INFO ] 2022-10-29 07:10:15,955 de.rwth.idsg.steve.ocpp.ws.WebSocketLogger - [chargeBoxId=Wall-E, sessionId=208f10a1-7595-7d0b-4424-494a9c372333] Received: [2,"37157","StatusNotification",{"connectorId":1,"errorCode":"NoError","status":"Available"}]                                                                                                                     
[INFO ] 2022-10-29 07:10:16,137 de.rwth.idsg.steve.ocpp.ws.WebSocketLogger - [chargeBoxId=Wall-E, sessionId=208f10a1-7595-7d0b-4424-494a9c372333] Sending: [3,"37157",{}]                                                                           
[INFO ] 2022-10-29 07:10:16,181 de.rwth.idsg.steve.ocpp.ws.WebSocketLogger - [chargeBoxId=Wall-E, sessionId=16198537-6497-7fdd-79aa-2106f44e33c9] Received: [2,"122659","StopTransaction",{"idTag":"00000000000000","meterStop":52022,"timestamp":"2022-10-29T07:10:15.868+00:00","transactionId":28}]                                                                        
[INFO ] 2022-10-29 07:10:16,183 de.rwth.idsg.steve.ocpp.ws.WebSocketLogger - [chargeBoxId=Wall-E, sessionId=208f10a1-7595-7d0b-4424-494a9c372333] Received: [2,"37158","StopTransaction",{"idTag":"00000000000000","meterStop":52022,"timestamp":"2022-10-29T07:10:15.868+00:00","transactionId":28}]                                                                         
[ERROR] 2022-10-29 07:10:16,428 de.rwth.idsg.steve.repository.impl.OcppServerRepositoryImpl - Exception occurred          
org.jooq.exception.DataAccessException: SQL [insert into `stevedb`.`transaction_stop` (`transaction_pk`, `event_timestamp`
, `event_actor`, `stop_timestamp`, `stop_value`, `stop_reason`) values (?, ?, ?, ?, ?, ?)]; Duplicate entry '28-2022-10-29
 07:10:16.000000' for key 'PRIMARY'                                                                                       
        at org.jooq_3.16.7.MYSQL.debug(Unknown Source) ~[?:?]                                                             
        at org.jooq.impl.Tools.translate(Tools.java:3095) ~[jooq-3.16.7.jar:?]                                            
        at org.jooq.impl.DefaultExecuteContext.sqlException(DefaultExecuteContext.java:670) ~[jooq-3.16.7.jar:?]          
        at org.jooq.impl.AbstractQuery.execute(AbstractQuery.java:354) ~[jooq-3.16.7.jar:?]                               
        at org.jooq.impl.AbstractDelegatingQuery.execute(AbstractDelegatingQuery.java:115) ~[jooq-3.16.7.jar:?]           
        at de.rwth.idsg.steve.repository.impl.OcppServerRepositoryImpl.updateTransaction(OcppServerRepositoryImpl.java:280) ~[steve.jar:?]                                                                                                          
        at de.rwth.idsg.steve.service.CentralSystemService16_Service.stopTransaction(CentralSystemService16_Service.java:221) ~[steve.jar:?]                                                                                                        
        at de.rwth.idsg.steve.ocpp.soap.CentralSystemService16_SoapServer.stopTransaction(CentralSystemService16_SoapServer.java:117) ~[steve.jar:?]                                                                                                
[... cut ...]
[INFO ] 2022-10-29 07:10:17,587 de.rwth.idsg.steve.ocpp.ws.WebSocketLogger - [chargeBoxId=Wall-E, sessionId=ea7f8986-0e64-bc9a-8052-35ecee81c321] Sending: [3,"1360",{"idTagInfo":{"status":"Accepted","expiryDate":"2022-10-29T08:10:17.517Z","parentIdTag":"default"}}]                                                                                                     
```

Why is event_timestamp part of the PK in transaction_stop? 

#### Additional context

Homecharge HC11L Wallbox, using [SECC01 Controller](https://www.efr.de/en/products/smart-energy-devices/control-box/secc-01-charging-controller/) from EFR.

goekay, 2022-10-31:
we try to let as much as possible succeed and to find a balance in strictness. 

the exceptions occur only if steve tries to insert both entries at the same event_timestamp (for this, you must be really unlucky, and the charging station must misbehave. unfortunately, this is one of your cases.) we aim to catch weird behavior or implementation issues of stations while keeping db clean and consistent.

otherwise, events with slightly varying event_timestamps will be persisted to db, because we want to allow the charging station to change its mind _later_ and send another stop transaction message. these subsequent updates would be persisted in db. these subsequent messages should not matter though, because we use the entry with the most recent timestamp to deduce the transaction stop values.

tvand, 2022-11-03:
Understood. I guess I will have to live with the charging station behavior for now. 
So I need to inflate my select statement a little bit. 
As the exceptions are logged to transaction_stop_failed, do I need to clean up that table occasionally?

goekay, 2022-11-03:
> As the exceptions are logged to transaction_stop_failed, do I need to clean up that table occasionally?

as you wish. it depends on how important historical data is for you and what you do with it. steve has no processes/flows/UIs depending on transaction_stop_failed. it's like a hidden feature for advanced users.

tvand, 2022-11-04:
Thank you for the clarification.