steve-community/steve #956: Add V1 baseline migration in order to avoid SUPER privilege need from 0.x schema version

juherr, 2022-11-05:
#### Checklist

- [x] I checked other issues already and found no answer
- [x] I checked the documentation and found no answer
- [x] I am running the latest version and the feature i am requesting is not implemented there

#### Describe the problem you are trying to solve

When you build Steve, you need to have a database with SUPER privilege otherwise you are not able to run migrations (see #495 ad other related issues).
This need was removed in a later migration and V1.x doesn't need it anymore.

#### Describe the solution you'd like

Flyway has baseline migration which skips migrations files when not needed.
https://flywaydb.org/documentation/tutorials/baselineMigrations
Having a V1.0 baseline migration will skip the migrations which need more privileges.

#### Describe alternative solutions or features you've considered 

Do it by hand:
1. Run migrations on a database with enough privilege
2. Export schema
3. Import schema into the target database without super privilege

#### Additional context

When working, it will be possible to remove the FAQ entry.
https://github.com/steve-community/steve/wiki/FAQ#database-migration-fails-on-mariadb-due-to-missing-super-privilege

goekay, 2026-02-27:
hey @juherr is this still relevant?

juherr, 2026-02-27:
It is supposed to be fixed by https://github.com/steve-community/steve/pull/1439