steve-community/steve #962: Firmware update does not work

kibeki, 2022-11-11:
#### Checklist

- [x ] I checked other issues already, but found no answer/solution
- [ x] I checked the documentation and wiki, but found no answer/solution
- [ x] I am running the latest version and the issue still occurs
- [ ] I am sure that this issue is about SteVe (and not about the charging station software or something unrelated to SteVe)

#### Specifications
```
SteVe Version     : 3.5.0
Operating system  :  5.4.0-131-generic #147-Ubuntu SMP
JDK               : OpenJDK Runtime Environment (build 11.0.17+8-post-Ubuntu-1ubuntu220.04)
Database          : mysql  Ver 15.1 Distrib 10.3.34-MariaDB
```

#### Expected Behavior
Mennekes Wallbox will download new firmware from the give the given URL and start to update.
...

#### Actual Behavior
We try to trigger the firmware update for Mennekes Amtrons Professional wallboxes. That does not work.
We don't see any traffice between the wallbox and Steve after perform that action in Steve.
...

#### Steps to Reproduce the Problem

  1. goto to Operations -> OCCP v1.6 -> Update Firmware
  2. select charge point from list
  3. give Location URI
  4. give Retrieve Date/Time

#### Additional context
...

lategoodbye, 2022-11-12:
@kibeki An extract of the Steve log would be helpful here.

kibeki, 2022-11-14:
Gave it a another try this morning and it works. Very happy. This makes it much easier to update our wallboxes.

[INFO ] 2022-11-14 09:00:00,673 de.rwth.idsg.steve.ocpp.ws.WebSocketLogger - [chargeBoxId=309, sessionId=ca10d1b3-6f86-3f1a-420b-931b8295a0a8] Received: [2, "44974712", "FirmwareStatusNotification", {"status": "Downloading"}]
[INFO ] 2022-11-14 09:01:29,895 de.rwth.idsg.steve.ocpp.ws.WebSocketLogger - [chargeBoxId=309, sessionId=ca10d1b3-6f86-3f1a-420b-931b8295a0a8] Received: [2, "47160276", "FirmwareStatusNotification", {"status": "Downloaded"}]

closed 2022-11-14 by kibeki

laradre, 2022-11-23:
@kibeki 

Hey, can you post an example of your location URI?

We are trying to update our mennekes amtron prof via steve but it seems like the wallbox cannot handle the command and I'm just guessing it could be bc of the URI.

Wallbox Log:
update_firmware_command(): URL = 'ftp://user:password@ip/download/ECU_APPS_5.12.6-11302.deb' 
xfer_parse_uri(): parsing URI 'ftp://user:password@ip/download/ECU_APPS_5.12.6-11302.deb' 
xfer_parse_uri(): Result: fileName 'ECU_APPS_5.12.6-11302.deb', host 'ip', host_path '/download', user 'user', password 'password' 
update_firmware_command(): URL = 'ftp://user:password@ip/download/ECU_APPS_5.12.6-11302.deb' 
update_firmware_command(): available disk space = '90340' kb 
ocpp_firmware_handle_update_command(): ADDR(ip) PATH(/download) 
update_firmware_command(): command handler failed

laradre, 2022-11-23:
> That works for the Amtron and Amedio chargepoints, but not for updating a Mennekes Gateways that we also run at two locations.

Unfortunately, it's not working for me, neither ftp nor http. Do you run your chargepoints in UTC timezone?

kibeki, 2022-11-23:
They have the same time.  Could it be a firewall issue. What shows the log of the Amtron while you try?

laradre, 2022-11-23:
We have looked into the firewall rules and did some test. Traffic passes from one to the other network.
The wallbox log really only says "could not handle command" but i can't tell why.

- websocket_msg_received len:174 = '[2,"41d28fb5-0aba-471c-bb1e-bd4ece590d84","UpdateFirmware",{"retrieveDate":"2022-11-23T14:00:00.000Z","location":"http://ip/ECU_APPS_5.12.6-11302.deb","retries":3}]' 
- websocket_msg_received(): COMMAND, processing... 
- json_comm_parse_ts(): failed to parse datetime '2022-11-23T14:00:00.000Z', retrying with '2022-11-23T14:00:00Z' 
- json_comm_parse_ts(): successfully parsed fixed datetime '2022-11-23T14:00:00Z' 
- websocket_msg_received(): successfully parsed COMMAND (14), notifying... 
- OCPP UpdateFirmware.req received [2,"41d28fb5-0aba-471c-bb1e-bd4ece590d84","UpdateFirmware",{"retrieveDate":"2022-11-23T14:00:00.000Z","location":"http://ip/ECU_APPS_5.12.6-11302.deb","retries":3}] 
- ocpp_server_handle_backend_command(): command 0x1edbd48 confirmation 0x1ed7d38 
- ocpp_server_handle_backend_command(): Received ocppUpdateFirmware 
- update_firmware_command(): URL = 'http://ip/ECU_APPS_5.12.6-11302.deb' 
- xfer_parse_uri(): parsing URI 'http://ip/ECU_APPS_5.12.6-11302.deb' 
- xfer_parse_uri(): Result: fileName 'ECU_APPS_5.12.6-11302.deb', host 'ip', host_path '', user '', password '' 
- update_firmware_command(): URL = 'ip' 
- update_firmware_command(): available disk space = '91492' kb 
- ocpp_firmware_handle_update_command(): ADDR(ip) PATH() 
- update_firmware_command(): command handler failed 
- ocpp_firmware_send_error_notification(): status: download failed 
- ocpp_client_send_firmware_status_notification(): status download failed 
- create_firmware_status_notification(): status: download failed 
- send_request_to_backend(): (OCPP FirmwareStatusNotification mode:queued persistent not transaction relevant) 
- OCPP FirmwareStatusNotification.req sent [2, "14367199", "FirmwareStatusNotification", {"status": "DownloadFailed"}] 
- error_handle_error_event(): Firmware update failed could not handle command 
- Firmware update failed could not handle command