steve-community/steve #963: Charging profile assignment is not properly removed

lategoodbye, 2022-11-11:
#### Checklist

- [x] I checked other issues already, but found no answer/solution
- [x] I checked the documentation and wiki, but found no answer/solution
- [x] I am running the latest version and the issue still occurs
- [x] I am sure that this issue is about SteVe (and not about the charging station software or something unrelated to SteVe)

#### Specifications
```
SteVe Version     : 3.4.9
Operating system  : Debian 11
JDK               : OpenJDK 17.0.4
Database          : Maria DB 10.5
```

#### Expected Behavior

In case SteVe gets a notification that a specific charging profile is not active anymore, the charging profile assignment to a specific charge point should be dropped.

#### Actual Behavior

In some corner cases the charge profile assignment persists and prevents modification on that charging profile.

de.rwth.idsg.steve.SteveException: Cannot modify this charging profile, since the following stations are still using it: ...

#### Steps to Reproduce the Problem

Corner case charging profile is cleared locally on the charging station:
  1. Install charging profile on Charge point via SetChargingProfile
  2. Remove charging profile locally
  3. Clear the charging profile on Charge point via ClearChargingProfile with ID
  4. Charge point returns "Unknown" but Steve keeps the assignment

goekay, 2022-11-20:
> In case SteVe gets a notification that a specific charging profile is not active anymore

when is this happening? and why?

lategoodbye, 2022-11-20:
Notification is a bad wording, i mean message. As discussed in the pull request in OCPP 1.6 there are only two types of response status (Accepted or Unknown).

goekay, 2022-11-23:
@lategoodbye can we close this issue since the PR is merged?

closed 2022-11-23 by lategoodbye