steve-community/steve #964: Unable to edit SteVe internal properties of charging profile while assigned to charge point

lategoodbye, 2022-11-11:
#### Checklist

- [x] I checked other issues already, but found no answer/solution
- [x] I checked the documentation and wiki, but found no answer/solution
- [x] I am running the latest version and the issue still occurs
- [x] I am sure that this issue is about SteVe (and not about the charging station software or something unrelated to SteVe)

#### Specifications
```
SteVe Version     : 3.4.9
Operating system  : Debian 11
JDK               : OpenJDK 17.0.4
Database          : Maria DB 10.5
```

#### Expected Behavior

It's possible to modify SteVe internal properties (description, additional note) while the charging profile is active on a charging station.

#### Actual Behavior

Currently SteVe complains with a exception `de.rwth.idsg.steve.SteveException: Cannot modify this charging profile, since the following stations are still using it`, but this is an unnecessary restriction because these properties are not sent to the charging station and this is not very userfriendly.

#### Steps to Reproduce the Problem

  1. Assign a charging station a new Charging profile
  2. Go to "Charging Profile Detail form"
  3. Modify description or additional note and press Update

goekay, 2022-11-20:
just checking: are description and note the only properties to remove the restriction? 

others seem to come directly from the charging profile ocpp data model.

lategoodbye, 2022-11-20:
Yes, correct.