steve-community/steve #969: OCPP server stopped automatically

subhankar007b, 2022-11-18:
#### Checklist

- [ ] I checked other issues already, but found no answer/solution
- [ ] I checked the documentation and wiki, but found no answer/solution
- [ ] I am running the latest version and the issue still occurs
- [ ] I am sure that this issue is about SteVe (and not about the charging station software or something unrelated to SteVe)

#### Specifications
```

#### Expected Behavior
OCPP Server should not be stopped until I do it manually.
...

#### Actual Behavior
OCPP Server stopped Automatically after usage of sometime
...

#### Steps to Reproduce the Problem

  1. ...
  2. ...
  3. ...

#### Additional context

...

goekay, 2022-11-20:
> OCPP Server stopped Automatically after usage of sometime

why? you did not provide any context, any log entries. nothing. people use it without the behaviour you experience. i have no reason to believe that the problem lies within steve.

closed 2022-11-20 by goekay

subhankar007b, 2022-11-21:
I didn't attach any log because there is no logs are there when it crashed. And you believe it or not, but its crashing after continuous usage.

subhankar007b, 2022-11-21:
I appreciate the work you have done. I didn't attach any log because there is no logs are there when it crashed. And its crashing after continuous usage.