steve-community/steve #982: AWS - Java heap space

sevelm, 2022-11-28:
Hy,
I just followed the instructions below:
https://github.com/steve-community/steve/wiki/Create-SteVe-Instance-in-AWS-Lightsail

unfortunately I get the following error:
`
[INFO] Changes detected - recompiling the module!
[INFO] Compiling 311 source files to /home/ubuntu/steve/target/classes
[INFO] ------------------------------------------------------------------------
[INFO] BUILD FAILURE
[INFO] ------------------------------------------------------------------------
[INFO] Total time:  03:04 min
[INFO] Finished at: 2022-11-28T08:58:32Z
[INFO] ------------------------------------------------------------------------
[ERROR] Java heap space -> [Help 1]
[ERROR]
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR]
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/OutOfMemoryError
`
Is this command in the instructions Still actual:
`
export MAVEN_OPTS="-Xmx100m" && mvn package
`

sevelm, 2022-12-01:
with the version 3.4.5 and compile with:
`export MAVEN_OPTS="-Xmx1024m" && mvn package`
did it work. However, I would also like to be up to date via AWS and it should also be possible to compile 3.5. unfortunately this one always breaks off.
anyone have experience or an idea how to get 3.5 running on an AWS server?
my JAVA knowledge is modest...

THX

keeth, 2023-01-14:
Maybe try not building the Steve jar on your cloud instance.  The maven build may have higher memory requirements than simply running the Steve server. Build the jar somewhere else and upload it to your cloud instance.

FWIW I am running both steve and mariadb on a 2GB instance.

goekay, 2024-02-18:
closing due to inactivity.

closed 2024-02-18 by goekay