steve-community/steve #983: Access from outside localhost

WueBumbum, 2022-12-02:
Hello,

I am new to SteVe and installed it on a fresh installed Raspberry Pi 4 according to the instructions here. It works from the pi @localhost. But as soon as I try to access the SteVe Web-Interface from another computer in my LAN I get an connection refused error.

I tried to find this issue here, but have not found an solution. As far as I understood SteVe is using jetty as webserver. According to my web research, jetty is configured by default to accept only connections from localhost. To change this I have to change the settings in /etc/default/jetty. (The setting should be JETTY_HOST=0.0.0.0)
But this configuration file does not exist on my installation. I found some jetty files in sub directiories of the SteVe build, but also have not found the configuration file there.

Am I on the correct track? How can I open access to SteVe outside localhost? Where can I find the jetty configuration used from SteVe?

best regards
Bumbum

goekay, 2022-12-02:
> To change this I have to change the settings in /etc/default/jetty. (The setting should be JETTY_HOST=0.0.0.0)

i wonder where you find these settings in steve because steve does not expose jetty settings.

[this](https://github.com/steve-community/steve/blob/master/src/main/resources/config/prod/main.properties#L22) is the config to change.

WueBumbum, 2022-12-05:
Hello goekay,

thank you for the hint. This has helped!

best regareds
Bumbum

closed 2022-12-05 by WueBumbum