steve-community/steve #989: package jooq.steve.db.XXX does not exist

iayun, 2022-12-08:
#### Checklist

- [ ] I checked other issues already, but found no answer/solution
- [ ] I checked the documentation and wiki, but found no answer/solution
- [ ] I am running the latest version and the issue still occurs
- [ ] I am sure that this issue is about SteVe (and not about the charging station software or something unrelated to SteVe)

#### Specifications
```
SteVe Version     : ...
Operating system  : ...
JDK               : ...
Database          : ...
```

#### Expected Behavior

...

#### Actual Behavior

...

#### Steps to Reproduce the Problem

  1. maven build
  2. ...
  3. ...

#### Additional context

...

goekay, 2022-12-08:
pls make sure that the build can access the database.

iayun, 2022-12-09:
[image]
not only database; is the package

iayun, 2022-12-09:
[image]
[image]

not only database; is the package

goekay, 2022-12-09:
pls search within the previous issues. we had similar questions.

closed 2022-12-10 by goekay