steve-community/steve #990: Could you add remote start and stop API endpoint?

ByungJun25, 2022-12-09:
#### Describe the problem you are trying to solve
Hi, Steve team! 
Thank you for creating good open-source.

Recently, your team merged the API endpoint for communicating with third party service. 
But unfortunately, there are no endpoints for remote starting and stopping.
I am trying to interact with steVe with a third party service, and I need remote starts and stops function.

#### Describe the solution you'd like
Please add more endpoint like below
* [ ] Remote start
* [ ] Remote stop

artemboyko43, 2023-04-01:
Any updates here? I guess this is one of most valueable features.

ByungJun25, 2023-04-11:
Hi @artemboyko43, You can get updates from #1000 issue which covers several API endpoints, including this issue.

juherr, 2023-05-30:
Duplicate of #388