steve-community/steve #1028: maven cant find generated joog classes

OzanAZamir, 2023-01-08:
#### Checklist

- [x] I checked other issues already, but found no answer/solution
- [x] I checked the documentation and wiki, but found no answer/solution
- [x] I am running the latest version and the issue still occurs
- [x] I am sure that this issue is about SteVe (and not about the charging station software or something unrelated to SteVe)

#### Specifications
```
SteVe Version     : 3.5.0
Operating system  : windows 10
JDK               : 11
Database          : MYSQL
```

#### Expected Behavior
Build succesfully
...

#### Actual Behavior
Compilation error
...

#### Steps to Reproduce the Problem

  I followed the steps in the readme for the build the project.
 joog classes generated at Path: C:\Users\ozan4\Documents\zamir\steve-master1\steve-master\target\generated-sources\jooq\steve\db

[ERROR] COMPILATION ERROR :
[INFO] -------------------------------------------------------------
[ERROR] /C:/Users/ozan4/Documents/zamir/steve-master1/steve-master/src/main/java/de/rwth/idsg/steve/repository/impl/OcppTagRepositoryImpl.java:[50,1] cannot find symbol
  symbol:   static OCPP_TAG_ACTIVITY
  location: class jooq.steve.db.tables.OcppTagActivity
[ERROR] /C:/Users/ozan4/Documents/zamir/steve-master1/steve-master/src/main/java/de/rwth/idsg/steve/repository/impl/ReservationRepositoryImpl.java:[47,1] cannot find symbol
  symbol:   static RESERVATION
  location: class jooq.steve.db.tables.Reservation
[ERROR] /C:/Users/ozan4/Documents/zamir/steve-master1/steve-master/src/main/java/de/rwth/idsg/steve/repository/impl/OcppServerRepositoryImpl.java:[57,1] cannot find symbol
  symbol:   static TRANSACTION_START
  location: class jooq.steve.db.tables.TransactionStart
[ERROR] /C:/Users/ozan4/Documents/zamir/steve-master1/steve-master/src/main/java/de/rwth/idsg/steve/repository/impl/OcppServerRepositoryImpl.java:[58,1] cannot find symbol
  symbol:   static TRANSACTION_STOP
  location: class jooq.steve.db.tables.TransactionStop
[ERROR] /C:/Users/ozan4/Documents/zamir/steve-master1/steve-master/src/main/java/de/rwth/idsg/steve/repository/impl/OcppServerRepositoryImpl.java:[59,1] cannot find symbol
  symbol:   static TRANSACTION_STOP_FAILED
  location: class jooq.steve.db.tables.TransactionStopFailed
[ERROR] /C:/Users/ozan4/Documents/zamir/steve-master1/steve-master/src/main/java/de/rwth/idsg/steve/repository/impl/SettingsRepositoryImpl.java:[40,1] cannot find symbol
  symbol:   static SETTINGS
  location: class jooq.steve.db.tables.Settings
[ERROR] /C:/Users/ozan4/Documents/zamir/steve-master1/steve-master/src/main/java/de/rwth/idsg/steve/repository/impl/GenericRepositoryImpl.java:[36,1] cannot find symbol
  symbol:   static RESERVATION
  location: class jooq.steve.db.Tables
[ERROR] /C:/Users/ozan4/Documents/zamir/steve-master1/steve-master/src/main/java/de/rwth/idsg/steve/repository/impl/GenericRepositoryImpl.java:[37,1] cannot find symbol
  symbol:   static TRANSACTION
  location: class jooq.steve.db.Tables
[ERROR] /C:/Users/ozan4/Documents/zamir/steve-master1/steve-master/src/main/java/de/rwth/idsg/steve/repository/impl/GenericRepositoryImpl.java:[40,1] cannot find symbol
  symbol:   static SCHEMA_VERSION
  location: class jooq.steve.db.tables.SchemaVersion
[ERROR] /C:/Users/ozan4/Documents/zamir/steve-master1/steve-master/src/main/java/de/rwth/idsg/steve/repository/impl/TransactionRepositoryImpl.java:[51,1] cannot find symbol
  symbol:   static TRANSACTION
  location: class jooq.steve.db.tables.Transaction
[ERROR] /C:/Users/ozan4/Documents/zamir/steve-master1/steve-master/src/main/java/de/rwth/idsg/steve/repository/impl/TransactionRepositoryImpl.java:[52,1] cannot find symbol
  symbol:   static TRANSACTION_START
  location: class jooq.steve.db.tables.TransactionStart
[ERROR] /C:/Users/ozan4/Documents/zamir/steve-master1/steve-master/src/main/java/de/rwth/idsg/steve/repository/impl/ChargingProfileRepositoryImpl.java:[49,1] cannot find symbol
  symbol:   static CHARGING_PROFILE
  location: class jooq.steve.db.Tables
[ERROR] /C:/Users/ozan4/Documents/zamir/steve-master1/steve-master/src/main/java/de/rwth/idsg/steve/repository/impl/ChargingProfileRepositoryImpl.java:[50,1] cannot find symbol
  symbol:   static CHARGING_SCHEDULE_PERIOD
  location: class jooq.steve.db.Tables
[ERROR] /C:/Users/ozan4/Documents/zamir/steve-master1/steve-master/src/main/java/de/rwth/idsg/steve/repository/impl/ChargingProfileRepositoryImpl.java:[52,1] cannot find symbol
  symbol:   static CONNECTOR_CHARGING_PROFILE
  location: class jooq.steve.db.Tables
[ERROR] /C:/Users/ozan4/Documents/zamir/steve-master1/steve-master/src/main/java/de/rwth/idsg/steve/utils/TransactionStopServiceHelper.java:[42,36] cannot find symbol
  symbol:   method getValue()
  location: variable v of type de.rwth.idsg.steve.repository.dto.TransactionDetails.MeterValues
[ERROR] /C:/Users/ozan4/Documents/zamir/steve-master1/steve-master/src/main/java/de/rwth/idsg/steve/utils/TransactionStopServiceHelper.java:[48,33] cannot find symbol
  symbol:   method getValue()
  location: variable v of type de.rwth.idsg.steve.repository.dto.TransactionDetails.MeterValues
[ERROR] /C:/Users/ozan4/Documents/zamir/steve-master1/steve-master/src/main/java/de/rwth/idsg/steve/utils/TransactionStopServiceHelper.java:[57,36] cannot find symbol
  symbol:   method getReadingContext()
  location: variable v of type de.rwth.idsg.steve.repository.dto.TransactionDetails.MeterValues
[ERROR] /C:/Users/ozan4/Documents/zamir/steve-master1/steve-master/src/main/java/de/rwth/idsg/steve/utils/TransactionStopServiceHelper.java:[58,39] cannot find symbol
  symbol:   method getFormat()
  location: variable v of type de.rwth.idsg.steve.repository.dto.TransactionDetails.MeterValues
[ERROR] /C:/Users/ozan4/Documents/zamir/steve-master1/steve-master/src/main/java/de/rwth/idsg/steve/utils/TransactionStopServiceHelper.java:[59,39] cannot find symbol
  symbol:   method getMeasurand()
  location: variable v of type de.rwth.idsg.steve.repository.dto.TransactionDetails.MeterValues
[ERROR] /C:/Users/ozan4/Documents/zamir/steve-master1/steve-master/src/main/java/de/rwth/idsg/steve/utils/TransactionStopServiceHelper.java:[60,39] cannot find symbol
  symbol:   method getLocation()
  location: variable v of type de.rwth.idsg.steve.repository.dto.TransactionDetails.MeterValues
[ERROR] /C:/Users/ozan4/Documents/zamir/steve-master1/steve-master/src/main/java/de/rwth/idsg/steve/utils/TransactionStopServiceHelper.java:[61,39] cannot find symbol
  symbol:   method getUnit()
  location: variable v of type de.rwth.idsg.steve.repository.dto.TransactionDetails.MeterValues
[ERROR] /C:/Users/ozan4/Documents/zamir/steve-master1/steve-master/src/main/java/de/rwth/idsg/steve/utils/TransactionStopServiceHelper.java:[62,39] cannot find symbol
  symbol:   method getPhase()
  location: variable v of type de.rwth.idsg.steve.repository.dto.TransactionDetails.MeterValues
...................................................................
but maven cant find generated class when build the project.

juherr, 2023-01-27:
Could you confirm that the jooq classes are well generated?
What is the maven command you run?

goekay, 2023-02-21:
closing due to inactivity.

closed 2023-02-21 by goekay