steve-community/steve #1029: [HELP] CORS ISSUE: API fetch not working on React JS Application - "401 Unauthorized Error"

PradeepGanapathi, 2023-01-11:
#### Checklist

- [ ] I checked other issues already, but found no answer/solution
- [ ] I checked the documentation and wiki, but found no answer/solution
- [ ] I am running the latest version and the issue still occurs
- [ ] I am sure that this issue is about SteVe (and not about the charging station software or something unrelated to SteVe)

#### Specifications
```
SteVe Version     : 3.5.0
Operating system  : windows 11
JDK               : 17.0.1
Database          : MySql 8.0.3
```

#### Expected Behavior
  -Expected to fetch the transaction details data using REST API and display it on the web page via React Web Application
...
#### Actual Behavior
   -Works perfectly with postman
   -But when tried using fetch in React App it shows CORS error.

[image]

   -Tried giving mode:"no-cors" in fetch call still it gives **401 unauthorized Error.**
[image]

[image]

...

#### Steps to Reproduce the Problem

  1. Get the REST API from backend.
  2. Do a fetch call to that API in a React App.

#### Additional context
We tried adding the response header in the TransactionsRestController class like below,
but still not working

public class TransactionsRestController {

 

    private final TransactionRepository transactionRepository;

 

    @ApiResponses(value = {
        @ApiResponse(code = 200, message = "OK", responseHeaders = {
          @ResponseHeader(
                    name = "Access-Control-Allow-Origin",
                    response = String.class,
                    description = "*"
                  ), 
          @ResponseHeader(
                    name = "Access-Control-Allow-Headers",
                    response = String.class,
                    description = "Origin, X-Requested-With, Content-Type, Accept, Authorization"
                  ),
          @ResponseHeader(
                    name = "Access-Control-Allow-Credentials",
                    response = String.class,
                    description = "true"
                  ), 
          @ResponseHeader(
                    name = "Access-Control-Allow-Methods",
                    response = String.class,
                    description = "GET,POST,PUT,DELETE,OPTIONS,HEAD"
                  )}),
        @ApiResponse(code = 400, message = "Bad Request", response = ApiErrorResponse.class),
        @ApiResponse(code = 401, message = "Unauthorized", response = ApiErrorResponse.class),
        @ApiResponse(code = 500, message = "Internal Server Error", response = ApiErrorResponse.class)}
    )
    @GetMapping(value = "")
   ........
   .......

...

juherr, 2023-01-11:
Steve doesn't provide CORS methods.

As a workaround, you can include your react views in a Next.js app where the backend will request steve and expose the expected api to your frontend.

keeth, 2023-01-14:
Adding a proxy could also help.

https://create-react-app.dev/docs/proxying-api-requests-in-development/

goekay, 2024-02-18:
old issue but wanted to give my 2cts before closing it:

the annotations you are adding are purely for documentation purposes in OpenAPI. they do not change the behaviour in code, since they are not spring annotations. this is the reason for not working.

i would suggest to look into the method body of `de.rwth.idsg.steve.config.SecurityConfiguration#apiKeyFilterChain` and modify the chain there as wanted. or, you could look into `org.springframework.web.bind.annotation.CrossOrigin`

closed 2024-02-18 by goekay