steve-community/steve #1042: Adding security headers for OCPP 1.5

01mk027, 2023-01-24:
Hi i need to add security headers into XML files of OCPP 1.5 for both request and response. How i can add? From where i must start?

juherr, 2023-01-27:
What do you refer to by "security header"?
As far as I know, security header is not out-of-the-box in SOAP and OCPP 1.5 doesn't describe any security mechanism.

Could you describe more about what you are trying to do?
What is the charging station (manufacturer, model, firmware version) you are trying to connect to Steve?

01mk027, 2023-01-27:
Actually, i need to see raw xml message which comes from charge station, and i want to control usernametoken (username and password) inside the header of relevant XML message. I'm focused on MediatorInInterceptor.java file to get raw XML, but i am still not successful.

goekay, 2023-01-27:
apache cxf (the soap lib we use) has the notion of interceptors with various phases. you can impl your own interceptor and register it in the chain at the appropriate place. depending on the place, the message is exposed to you in different forms: raw stream, xml, marshalled request etc. you can control the behaviour of the chain within your interceptor.

https://cxf.apache.org/docs/interceptors.html

01mk027, 2023-02-03:
i searched for interceptors, and i added interceptor to OCCPConfiguration.java, and i achieved to enable username password control in xml headers. now, i try to find  where soap responses sent to charge point, and i need to retrieve soap responses as XML. Does anybody know?

closed 2024-09-20 by goekay