steve-community/steve #1043: Unable to authenticate with API

cbrandlehner, 2023-01-24:
#### Checklist

- [X] I checked other issues already, but found no answer/solution
- [X] I checked the documentation and wiki, but found no answer/solution
- [X] I am running the latest version and the issue still occurs
- [X] I am sure that this issue is about SteVe (and not about the charging station software or something unrelated to SteVe)

#### Specifications
```
SteVe Version     : 3.5.0
Operating system  : Dockerfile: FROM maven:3.6.1-jdk-11
JDK               : Dockerfile: FROM maven:3.6.1-jdk-11
Database          : mariadb:10.4
```

#### Expected Behavior

I would like to to use the new API documented under:
/steve/manager/v3/api-docs

#### Actual Behavior
Whatever I try, the API always returns 401 "Full authentication is required to access this resource"

#### Steps to Reproduce the Problem

Configured STEVE using the new keys:
```
webapi.key = STEVE-API-KEY
webapi.value = 12345
```

Created a CURL command to access the API. The URL used is one of the URLs published by "/steve/manager/v3/api-docs".  

```
curl -i \
  -H "Content-Type: application/json" \
  -d '
{ "auth": {
    "identity": {
      "methods": ["password"],
      "password": {
        "user": {
          "name": "STEVE-API-KEY",
          "password": "12345"
        }
      }
    }
  }
}' \
  "http://steve.home.arpa:8180/steve/api/v1/ocppTags" ; echo
```

I also tried by just accessing the URL in the browser, but with the same result. There is no login prompt.

juherr, 2023-01-27:
It looks like the openapi description is not good with auth.

But the documentation will help: https://github.com/steve-community/steve/wiki/Configuration#rest-apis
It refers to https://github.com/steve-community/steve/blob/master/src/main/resources/config/prod/main.properties#L20-L24  where you'll see that you need to use a header.

According to your sample, following should work:
```
curl -H "STEVE-API-KEY: 12345" http://steve.home.arpa:8180/steve/api/v1/ocppTags
```

goekay, 2023-01-27:
thanks for bringing this to my attention. i will add the curl example to wiki docs.

cbrandlehner, 2023-01-27:
I confirm that the curl example posted by @juherr works.

Thanks for your effort. Good job!

juherr, 2023-01-27:
@goekay In fact the `securitySchemes` and the `securityContexts` is missing from the docket description: https://github.com/steve-community/steve/blob/faf7c166ce6011adf30a28e8f2dfbdd954b69003/src/main/java/de/rwth/idsg/steve/config/ApiDocsConfiguration.java#L61-L66

You can find a sample where it is configured here: https://springfox.github.io/springfox/docs/current/#springfox-spring-mvc-and-spring-boot

juherr, 2023-01-30:
@goekay Didn't test yet but the change is available here: https://github.com/juherr/steve/commit/9d11192161c7faad7eb818a77a26deb7145cded5

goekay, 2024-08-08:
thanks @juherr for your proposed change. the analogous implementation for swagger v3 can be reviewed in referenced PR.

closed 2024-08-09 by goekay