steve-community/steve #1045: API date problem

cbrandlehner, 2023-01-27:
#### Checklist

- [X] I checked other issues already, but found no answer/solution
- [X] I checked the documentation and wiki, but found no answer/solution
- [X] I am running the latest version and the issue still occurs
- [X] I am sure that this issue is about SteVe (and not about the charging station software or something unrelated to SteVe)

#### Specifications
```
SteVe Version     : 3.5.0
Operating system  : Dockerfile: FROM maven:3.6.1-jdk-11
JDK               : Dockerfile: FROM maven:3.6.1-jdk-11
Database          : mariadb:10.4
```

#### Expected Behavior

I would like to see all transactions after a given date/time using the REST API.

#### Actual Behavior

API always returns a list of all transactions

#### Steps to Reproduce the Problem

Get list of transactions for a given tag.

`curl -H "STEVE-API-KEY: 12345" http://steve.home.arpa:8180/steve/api/v1/transactions\?ocppIdTag\=12345`

That curl returns this list:
`{"id":12,"connectorId":1,"chargeBoxPk":2,"ocppTagPk":8,"chargeBoxId":"chargePointSim","ocppIdTag":"12345","startValue":"0","startTimestamp":"2022-11-29T19:26:43.000Z","stopValue":"52","stopReason":"EVDisconnected","stopTimestamp":"2022-11-29T19:27:09.000Z","stopEventActor":"station"},{"id":11,"connectorId":1,"chargeBoxPk":2,"ocppTagPk":8,"chargeBoxId":"chargePointSim","ocppIdTag":"12345","startValue":"0","startTimestamp":"2022-11-29T18:28:47.000Z","stopValue":"52","stopReason":"EVDisconnected","stopTimestamp":"2022-11-29T18:29:13.000Z","stopEventActor":"station"},{"id":7,"connectorId":1,"chargeBoxPk":2,"ocppTagPk":8,"chargeBoxId":"chargePointSim","ocppIdTag":"12345","startValue":"0","startTimestamp":"2022-11-23T19:37:48.000Z","stopValue":"52","stopReason":"EVDisconnected","stopTimestamp":"2022-11-23T19:38:13.000Z","stopEventActor":"station"},{"id":6,"connectorId":1,"chargeBoxPk":2,"ocppTagPk":8,"chargeBoxId":"chargePointSim","ocppIdTag":"12345","startValue":"0","startTimestamp":"2022-11-23T19:26:22.000Z","stopValue":"52","stopReason":"EVDisconnected","stopTimestamp":"2022-11-23T19:26:47.000Z","stopEventActor":"station"},{"id":5,"connectorId":1,"chargeBoxPk":2,"ocppTagPk":8,"chargeBoxId":"chargePointSim","ocppIdTag":"12345","startValue":"0","startTimestamp":"2022-11-23T19:21:19.000Z","stopValue":"52","stopReason":"EVDisconnected","stopTimestamp":"2022-11-23T19:21:44.000Z","stopEventActor":"station"}`

Now I would like to see all transactions since start of January. To do so I use the same time format as in the response above.
`curl -H "STEVE-API-KEY: 12345" http://steve.home.arpa:8180/steve/api/v1/transactions?from=2023-01-01T00:00:00.000Z`
throws an error:
`{"timestamp":"2023-01-27T11:46:27.015Z","status":400,"error":"Bad Request","message":"Error understanding the request","path":"http://steve.home.arpa:8180/steve/api/v1/transactions"}`

However, If I use a slightly different time format, I get a response:
`curl -H "STEVE-API-KEY: 12345" http://steve.home.arpa:8180/steve/api/v1/transactions\?from\=2023-01-01T00:00:00`
The response is a full list of all transactions (which I do not post here as it is a very long list). The list does NOT start at 2023-01-01 as expected.

goekay, 2023-01-27:
in the sample that results in an exception, you are using UTC. the api expects "ISO8601 without timezone". [reference in code](https://github.com/steve-community/steve/blob/master/src/main/java/de/rwth/idsg/steve/web/dto/QueryForm.java#L46-L50). this is documented in swagger api docs as well.

the slightly different format uses a query param that is not part of the schema and is therefore ignored. and hence the full list.

cbrandlehner, 2023-01-27:
@goekay I agree on your first statement. I used that example because that was the code auto-generated by swagger after reading the steve openapi specification.
This is the openapi spec part:
```
{
                        "name": "from",
                        "in": "query",
                        "description": "Show results that happened after this date/time. Format: ISO8601 without timezone. Example: `2022-10-10T09:00:00`",
                        "required": false,
                        "style": "form",
                        "schema": {
                            "type": "string",
                            "format": "date-time"
                        }
                    },
```
[Swagger specifies](https://swagger.io/docs/specification/data-models/data-types/) that the "date-time" format is based on [RFC3339](https://www.rfc-editor.org/rfc/rfc3339#section-5.6).

However, I do not understand why the second request, using ISO8601 does not meet the schema. Would you be so kind to privide an example that returns all transactions after a given point in time?

goekay, 2023-01-27:
the way you declare the parameter "from" with extra \ is unexpected in the second request. `that` is the schema violation, and not the date time value. therefore, steve cannot map this param to something and ignores it.

i did not test it, but the following should work:

`curl -H "STEVE-API-KEY: 12345" http://steve.home.arpa:8180/steve/api/v1/transactions?from=2023-01-01T00:00:00`

cbrandlehner, 2023-01-27:
That extra \ was an attempt to escape the question mark. My current request looks like this:

`curl -H "STEVE-API-KEY: 12345" "http://steve.home.arpa:8180/steve/api/v1/transactions?from=2023-01-01T00:00:00"`

but it still returns a full list of all transactions.

goekay, 2023-01-27:
oh, apologies for misleading you. my mistake. apparently, i forgot about one nuance when using the API: it mimics the query behaviour/possibilities and data representation of the web pages.

when you go the transactions web page, you will see a query param called [period type](https://github.com/steve-community/steve/blob/master/src/main/java/de/rwth/idsg/steve/web/dto/TransactionQueryForm.java#L49-L50). it takes [these possible values](https://github.com/steve-community/steve/blob/master/src/main/java/de/rwth/idsg/steve/web/dto/TransactionQueryForm.java#L94-L99). `from` and `to` are respected only when `periodType=FROM_TO`. and you have to set both (and not only `from` as in your example).

the same constraints apply to API calls. since your exemplary request does not set `periodType`, it defaults to `ALL`. in short: you have set `periodType=FROM_TO` and `from=...` and `to=...` in order to achieve what you want to achieve.

cbrandlehner, 2023-01-27:
Excellent, that did it.
I will update the wiki with an example.

closed 2023-01-27 by cbrandlehner