steve-community/steve #1046: how to do it right / initial installation

aRneee76, 2023-01-29:
Hello,

I hope someone can help me or give me a tip on how to pair my 8 Wallbe boxes with Steve. Steve is freshly installed, mail etc configured and all devices added. The on the web interface of the devices I have stored the ws address, as it is described in the instructions. I have tried two different notations. 
steve/websocket/CentralSystemService/
steve/websocket/CentralSystemService/Box-ID 

Unfortunately, the boxes do not appear in Steve Management. Can someone please give me a hint what I am either doing wrong or how to set it up correctly? 
[image]

Grateful
Arne

juherr, 2023-01-29:
Your `Wallbe` station is not referenced in the supported list: https://github.com/steve-community/steve/wiki/Charging-Station-Compatibility
Maybe you are the first to deal with it.
Could you provide the full description of the charging station (manufacturer, model, firmware version, and maybe a link to the documentation)?

Maybe the steves url is not the good one.
If `steve/websocket/CentralSystemService/` is the websocket path, what is the manager path?

Are you able to read the station logs and check what is going bad?

aRneee76, 2023-01-29:
Hello Julian,

thank you very much for the feedback. These are Wallbe Pro Plus devices with a SL-01.04.21 firmware.
The network interface (charge control) is a
EV-CC-AC1-M3-CBC-RCM-ETH

Described here:
https://ledwerkstatt.ch/wp-content/uploads/2020/05/um_de_ev_cc_ac1_m3_cbc_rcm_eth_108191_de_04-1.pdf

I have access to the log files, but the date and time is not correct. 
"1970-09-04T22:57:58.000Z";EVSE State: B
"1970-09-04T22:57:58.001Z";EVSE State: RCMB_TEST
"1970-09-04T22:58:01.000Z";EVSE State: CAR_WAKEUP
"1970-09-04T22:58:05.000Z";EVSE State: UNLOCKING
"1970-09-04T22:58:05.001Z";PWM Active: False
"1970-09-04T22:58:06.000Z";EVSE State: A
"1970-09-06T16:25:40.000Z";EVSE State: LOCKING
"1970-09-06T16:25:42.000Z";EVSE State: B
"1970-09-06T16:26:01.000Z";EVSE State: RCMB_TEST
"1970-09-06T16:26:01.001Z";PWM Active: True
"1970-09-06T16:26:04.000Z";EVSE State: CAR_WAKEUP
"1970-09-06T16:26:04.001Z";EVSE State: LOADING
"1970-09-06T19:08:52.000Z";EVSE State: B
"1970-09-06T19:08:52.001Z";EVSE State: RCMB_TEST
"1970-09-06T19:08:55.000Z";EVSE State: CAR_WAKEUP
"1970-09-07T00:18:15.000Z";EVSE State: LOADING
"1970-09-07T00:18:18.000Z";EVSE State: B
"1970-09-07T00:18:19.000Z";EVSE State: RCMB_TEST
"1970-09-07T00:18:21.000Z";EVSE State: CAR_WAKEUP
"1970-09-07T00:18:28.000Z";EVSE State: UNLOCKING
"1970-09-07T00:18:28.001Z";PWM Active: False
"1970-09-07T00:18:29.000Z";EVSE State: A
Boot;DIP_Position:0;Pot_Position:6
"1970-03-24T22:11:12.000Z";MODEM;Modem power on
Boot;DIP_Position:0;Pot_Position:6
"1970-01-01T00:00:03.000Z";MODEM;Modem power on

The Manager ist on
http://steve.domain.local:8080/steve/manager/home

goekay, 2023-01-29:
hey @aRneee76 the url seems to be alright. in case of failing websocket connection attempts, you should see some error logs in steve's logs. can you pls post them?

aRneee76, 2023-01-31:
Hello,

I have managed to communicate thanks to your help. Thanks a lot! It is, as it says in the screenshot, a mobile communication device that can only communicate via an APN (sim card) out. 
Who can read is clearly at an advantage.
Wallbe PRO Plus EV Charge Control can be added to the list of supported devices.
Thanks
Arne

closed 2023-01-31 by aRneee76

juherr, 2023-02-01:
Station added in the list