steve-community/steve #1054: Unable to add a charge point using curl

Chabanee, 2023-02-10:
#### Checklist

- [x] I checked other issues already, but found no answer/solution
- [x] I checked the documentation and wiki, but found no answer/solution
- [x] I am running the latest version and the issue still occurs
- [x] I am sure that this issue is about SteVe (and not about the charging station software or something unrelated to SteVe)

#### Specifications
```
SteVe Version     : 3.5.0
Operating system  : linux
JDK               : ...
Database          : ...
```

#### Expected Behavior

Give a solution to add a chargepoint using curl (ps: I'm using php).

#### Actual Behavior

Nothing

#### Steps to Reproduce the Problem

  1. Using a syntax to add a chargepoint with my script php or terminal. 
  2. Get no response.

#### Additional context

I'm also wondering what is the meanning of "dquote>" if you can help me.

[image]

juherr, 2023-02-11:
There is no api for that. See #1000

Chabanee, 2023-02-14:
Thanks for your answer,

but is there any way else to do that in php ?

juherr, 2023-02-14:
Yes. Not recommended but you can use pdo, connect to the database and add the expected entries in the appropriate tables.

Chabanee, 2023-02-14:
If you can explain me how please it will be really helpeful.

Thanks

juherr, 2023-02-14:
If you don't know, it will be too complicate.
Just wait for the expected api and you'll be able to use curl then.

Chabanee, 2023-02-14:
Alright, thank you.

Last question can I use php to connect to the database and add the expected entries in the appropriate tables ?

Chabanee, 2023-02-15:
oK thanks

closed 2023-02-21 by goekay