steve-community/steve #1059: I am using math -x OCPP1.6 library. I'm having trouble connecting. Any help would be appreciated.

jpily, 2023-02-16:
Thank you for providing the test tool for free.
I am using the math-x Arduino OCPP1.6 library and making several attempts to test the STeve server as a central system.

I successfully built the Steve server environment and it worked normally until build and run.

To test the math-x Arduino OCPP1.6 library, I made a charge box ID on the Steve server, and then set the math-x arduino ocpp library as follows.
<Arduino IDE code>
#define OCPP_HOST "127.0.0.1"
#define OCPP_PORT 9000
#define OCPP_URL "ws://127.0.0.1:9000/steve/websocket/CentralSystemService/test01"

The result of the operation is [AO] info (.pio/libdeps/esp32-s3-devkitc-1/ArduinoOcpp/src/ArduinoOcpp/Core/OcppSocket.cpp:31): Disconnected Connection failed with the log that the socket communication connection is not made. there is.

For reference, I did the following when creating a charge box id in Setve server.
1. menu dataManagement -> Charge point click -> click add New button -> enter add charge point contents

-----Input contents ------
<OCPP>
ChargeBox ID: test01
Insert connector status after start/stop transaction: check
Insert connector status after start/stop transaction: Accepted
<Address>
Street:test01
House Number :1234
zip code : 123456
city : seoul
Country: Korea
<Misc>
Description: dummey charge point
Admin Address: www.test.com
Latitude: 90
Longitude::90
------------------------------------------

I entered this and clicked the Add button to add it.
STeve server main.properties is set as follows:
-------------------------------------------------- -------------
context.path = steve

# Database configuration
#
db.ip = localhost
db.port = 3306
db.schema = stevedb
db.user = steve
db.password = tech7975

# Credentials for Web interface access
#
auth.user = admin
auth.password = 1234

# The header key and value for Web API access using API key authorization.
# Both must be set for Web APIs to be enabled. Otherwise, we will block all calls.
#
webapi.key = STEVE-API-KEY
webapi.value = 1234

# Jetty configuration
#
server.host = 127.0.0.1
server.gzip.enabled = true

# Jetty HTTP configuration
#
http.enabled = true
http.port = 9000

# Jetty HTTPS configuration
#
https.enabled = false
https.port = 8443
keystore.path =
keystore.password =

# When the WebSocket/Json charge point opens more than one WebSocket connection,
# we need a mechanism/strategy to select one of them for outgoing requests.
# For allowed values see de.rwth.idsg.steve.ocpp.ws.custom.WsSessionSelectStrategyEnum.
#
ws.session.select.strategy = ALWAYS_LAST

# if BootNotification messages arrive (SOAP) or WebSocket connection attempts are made (JSON) from unknown charging
# stations, we reject these charging stations, because stations with these chargeBoxIds were NOT inserted into database
#beforehand. by setting this property to true, this behavior can be modified to automatically insert unknown
# stations into database and accept their requests.
#
# CAUTION: setting this property to true is very dangerous, because we will accept EVERY BootNotification or WebSocket
# connection attempt from ANY sender as long as the sender knows the URL and sends a valid message.
#
auto.register.unknown.stations = false

### DO NOT MODIFY ###
steve.version = ${project.version}
git.describe = ${git.commit.id.describe}
db.sql.logging = false
profile=prod
-------------------------------------------------- -------------------------------------------------- ---

Any help would be appreciated.

phken91, 2023-03-20:
hi @jpily , i facing same issue with you, can you share your current working config here? thanks

jpily, 2023-03-20:
Does Steve server work properly?
You can do it the way I debugged it and made it work.
Please meet the prerequisites to properly use the arduinoOCpp library.
1. mysql local time command setting (whether it was executed once before, once again... unconditionally
    do)
     SET GLOBAL time_zone = '+00:00';

2. Connect to the Stever server and create a virtual charging point.
      Steve server start-> menu Data management->charge point
->add new -> Create ChargeBox ID

3. ArduinoOcpp code host, port, url setting
ex
#define STASSID "MySSID"
#define STAPSK "Ap_password"

#define OCPP_HOST "220.110.22.35" // Set the IP given when connecting from outside, not the local IP
#define OCPP_PORT 9000 // Steve server port
#define OCPP_URL "ws://220.110.22.35:9000/steve/websocket/CentralSystemService/test01"

If it does not work after following the above steps, the port is not activated.
If not, please contact us again. It's hard to answer because I'm at work.

phken91, 2023-03-20:
thanks for reply, i fixed this, by using [simulator](https://github.com/zzerk/OCPP-ChargePoint-Simulator) to help up debug

jpily, 2023-03-20:
Did you check the meter value with arduino Occpp? Check on Steve server...

rathourcode, 2023-04-19:
> Did you check the meter value with arduino Occpp? Check on Steve server...
> 
> If you don't mind, could you share the entire code?

HAI Jpily,
were you able to connect to websocket connection, if yes please help as i am unable to  acess please help

jpily, 2023-04-19:
I succeeded in connecting by doing the test myself

rathourcode, 2023-04-19:
Can you please tell the steps you followed

jpily, 2023-04-19:
This is the first part to check. Please check first and tell me
1. Check if the port you want to connect to is in the listening state

Run command prompt (cmd) -> netstat -ano Enter 127.0.0.1:port listening check

rathourcode, 2023-04-19:
sorry for late reply my connection got interrupted
[image]

jpily, 2023-04-19:
Have you finished installing Steve server?

rathourcode, 2023-04-19:
yes

> Have you finished installing Steve server?

yes

rathourcode, 2023-04-19:
> Have you finished installing Steve server? I'd like to know if Steve's server is running normally

yes i have instaleed steve sucessfuly
[image]

rathourcode, 2023-04-19:
i am able to acess steve  but when i try to run mathews library i get the same error as you posted cpp/31/disconeected

jpily, 2023-04-19:
Tell me the WS address and port number that will come out after Steve server starts

rathourcode, 2023-04-19:
> Tell me the WS address and port number that will come out after Steve server starts

ws://127.0.0.1:8080/steve/websocket/CentralSystemService/(chargeBoxId)

jpily, 2023-04-19:
Tell me the local ip

1. You must create a chargeBoxId on the Steve server.
2. After making it, please let me know chargeBoxId and local ip. I'll fix the code that works accordingly and upload it

rathourcode, 2023-04-19:
> Tell me the local ip
> 
> 1. You must create a chargeBoxId on the Steve server.
> 2. After making it, please let me know chargeBoxId and local ip. I'll fix the code that works accordingly and upload it

yes i i have created  an id  by name test1

jpily, 2023-04-19:
\syslog-master\src\main\cisco\config\dev main.properties
# Jetty HTTP settings
http.enabled = true
Request to modify-> http.port = 8080

jpily, 2023-04-19:
What is the internal ip? local ip is required

jpily, 2023-04-19:
Please understand that there is Korean because I am Korean

#include <Arduino.h>
#include <WiFi.h>
#include <ArduinoOcpp.h>

#define STASSID "your  wifi ID"
#define STAPSK  "whfi pass"

#define OCPP_HOST "192.168.1.103"
#define OCPP_PORT 8080
#define OCPP_URL "ws://192.168.1.103:8080/steve/websocket/CentralSystemService/test1"

const int LED_PIN = 16;
const int PWM_RANGE = 255;
const int PWM_STEP = 5;
const int BLINK_INTERVAL = 500;

void setup() {

  Serial.begin(115200);
  
  Serial.println("Wi-Fi 연결 중...");
  WiFi.begin(STASSID, STAPSK);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  
  OCPP_initialize(OCPP_HOST, OCPP_PORT, OCPP_URL);

  setConnectorPluggedInput([](){
      return true;
  });
  
  bootNotification("니즈로봇 충전소(공릉동)", "니즈로봇");
}

void loop() {
  OCPP_loop();
  
  if (!getTransactionIdTag()) {
    authorize("0123456789ABCD", [](JsonObject response){
      if (response["idTagInfo"]["status"] == "Accepted") {
        auto ret = beginTransaction("0123456789ABCD");
        if (ret) {
          Serial.println("[main] 거래 시작됨. ConnectorPlugged Input이 true가 되면 StartTransaction이 전송됩니다.");
        } else {
          Serial.println("[main] 거래 시작 실패.");
        }
      } else {
        Serial.printf("[main] Authorize 거부됨. 이유: %s\n", response["idTagInfo"]["status"] | "");
      }
    });
  }
  
  if (getTransactionIdTag()) {
    Serial.println("[main] 거래 진행 중. EV 플러그 비축화를 기다리는 중...");
    while (ocppPermitsCharge()) {
      OCPP_loop();
  

            // 서서히 밝아졌다가 작아지는 LED 제어
      for (int i = 0; i <= PWM_RANGE; i += PWM_STEP) {
        analogWrite(LED_PIN, i);
        delay(50);
      }
      for (int i = PWM_RANGE; i >= 0; i -= PWM_STEP) {
        analogWrite(LED_PIN, i);
        delay(50);
      }
      delay(100);
    }
    Serial.println("[main] EV 플러그 비축화 감지됨. 거래 종료...");
    endTransaction();
    Serial.println("[main] 거래 종료됨.");
  }

  delay(100);
}

jpily, 2023-04-19:
Please tell me the result. If it doesn't work, you'll have to move on

rathourcode, 2023-04-21:
hai jpily i tried to run your code but it shows Replace cached operation

i removed all earlier cached data but it still gives me this error , can you help

rathourcode, 2023-04-21:
were you able to acheive hearbeat using esp32 on steve ?

jpily, 2023-04-21:
If the development environment is Platformio, you can enable debugging mode to view the data transmission and reception process on the serial port

jpily, 2023-04-21:
I'm at work.. There's no way I can help you if it's not this time

goekay, 2023-08-29:
closing due to inactivity. i hope you resolved your issues after this exchange :)

closed 2023-08-29 by goekay