steve-community/steve #1061: API not reachable after setting key and value (404)

Seitenwerk, 2023-02-16:
#### Checklist

- [X] I checked other issues already, but found no answer/solution
- [X] I checked the documentation and wiki, but found no answer/solution
- [X] I am running the latest version and the issue still occurs
- [X] I am sure that this issue is about SteVe (and not about the charging station software or something unrelated to SteVe)

#### Specifications
```
SteVe Version     : 3.5.0
Operating system  : Ubuntu 20.04 LTS
JDK               : 11.0.17
Database          : 10.3.37-MariaDB
```

#### Expected Behavior

After setting API Key and Value in the prod main.properties file, compiling SteVe and running it again, expectation is that the API is exposed under documented urls

#### Actual Behavior

Requests to any of the available api related urls always return 404:
Example:

URI: | /steve/manager/v3/api-docs
404
Not Found
spring-dispatcher

#### Steps to Reproduce the Problem

  1. configure API key and value as documented
  2. compile SteVe
  3. run it
  4. Attempt requests to api urls or attempt to retrieve API scheme from specific url

#### Additional context

...

emerout, 2023-02-26:
@Seitenwerk you have to run Steve in prod mode in order to access api-docs:

```
# main.properties
profile = prod
```

see class: 
```java
@Configuration
@EnableOpenApi
@Conditional(SteveProdCondition.class)
public class ApiDocsConfiguration {

    static {
        // Set the path with prefix /manager to protect the documentation behind regular sign-in
        // Default is just /v3/api-docs
        System.setProperty("springfox.documentation.open-api.v3.path", "/manager/v3/api-docs");
    }

    ...
```

drocx, 2023-02-28:
I have the same problem.
API Key and API Value are set. Prod Mode is also set.
Could it be because there is no ApiDocsConfiguration.java in the release? This is only available in GitHub.

emerout, 2023-02-28:
@drocx 

> Could it be because there is no ApiDocsConfiguration.java in the release? This is only available in GitHub.

For sure, this feature is for now only in master branch.

You can wait for new release, or build steve from master branch.

drocx, 2023-02-28:
Now it works after we have build the version from master.

emerout, 2023-02-28:
@drocx Great, I think you could close the issue if all is fine.

closed 2023-08-20 by goekay