steve-community/steve #1065: [Question] User Frontend

tzuzn, 2023-02-22:
Hi,
is there a User Frontend for making reservations and Remote-Start/Remote-Stop Transactions?
If not is there anything planned in the near future or should i wait on #1000 and use the API to develope my own Frontend.
So far i've just developed a website which reads the DB to show which connectors are Availiable, but i dont feel comfortable directy writing into the DB.

Best Regards,
tzuzn

goekay, 2023-02-22:
> is there a User Frontend for making reservations and Remote-Start/Remote-Stop Transactions?

yes, there is. did you use the web ui of steve?

tzuzn, 2023-02-22:
I currently only use the Management Frontent of steve.

goekay, 2023-02-22:
all web ui possibilities are behind the login of a user. this one user, an admin user if you will, can interact with all offerings of steve. so, how are you using the management frontend without having logged in with this user? 

> As far as i've seen there is no way for users to login, that would be a requirement I had.

therefore, i cannot understand this claim.

> is there a User Frontend for making reservations and Remote-Start/Remote-Stop Transactions?

you can trigger/execute all ocpp operations, including the ones you listed, via the web ui after the user login. i suggest exploring the possibilities of web ui.

tzuzn, 2023-02-22:
of course i have logged in as admin to do these kind of operations, goal is to have users log in, and only be able to make reservations and start/stop remote transactions of their assigned Ocpp ID Tag

goekay, 2023-02-22:
ok, now we arrived at a very different description of your issue. you want your charging station clients/customers (not the user in the sense of technical/admin users of the web ui). this does not follow from your original comment.

there is no possibility for that. steve is not meant to be directly used by end users (i.e. clients/customers). it can or should be used by operators (CPOs: Charge Point Operators).

tzuzn, 2023-02-22:
so if I would want and web ui for end users I had to develop that myself. In my understanding the API-Endpoint should allow that in the future.

juherr, 2023-02-22:
#1000 or #991 are options to solve the request.

@tzuzn You won't be able to generate a request by inserting it into the database. The only workaround is using the UI by code with [WebDriver](https://www.selenium.dev/documentation/webdriver/) or another similar tool.

goekay, 2023-02-22:
@tzuzn i would not do something customer-facing with the current state of the APIs. once again, there is one technical user that can do all. permission control per user/customer is not defined. #991 would be a prerequisite for such a functionality.

goekay, 2023-02-22:
> @tzuzn i would not do something customer-facing with the current state of the APIs. once again, there is one technical user that can do all. permission control per user/customer is not defined. #991 would be a prerequisite for such a functionality.

... or your proxy service (that you would be developing) that is in front of steve handles all this customer-facing authentication/authorization and calls steve with the technical web API user.

tzuzn, 2023-02-22:
I think i should be able to make the permission control myself reading the database but making the actual reservations or transactions i guess i'll have to wait untill the APIs are ready.
thank you for your time 

> ... or your proxy service (that you would be developing) that is in front of steve handles all this customer-facing authentication/authorization and calls steve with the technical web API user.

yes i think that would be the best solution. tho i'd like to wait untill the APIs are ready respectively the reservations and transaction endpoints are implemented.

Thank you for your time.
looking forward on the development ov steve

closed 2023-02-22 by tzuzn