steve-community/steve #1066: I am looking for java file, where transactions can be done

01mk027, 2023-02-22:
Hi all;
I am currently trying to write a java file, which interacts with Steve's methods which will be triggered from outside. For example, if some authorized person make post request to http://127.0.0.1:8080/steve/api/v1/transactions15/datatransfer, i want to operate related transaction. i am looking for this puropose to files under service folder (it doesn't matter for OCPP 1.5 or OCPP1.6) , but i am not able to find exact file. Does anybody know how to be done?

juherr, 2023-02-22:
Hi,

What do you mean by "operate related transaction"? What is the purpose?

01mk027, 2023-02-22:
when an admin enter to steve from the address of http://127.0.0.1:8080/steve/manager/home, and all transactions can be selected and done, in the address of http://127.0.0.1:8080/steve/manager/operations/v1.5/{transactionName} (e.g ChangeAvailability). i want to operate these transactions without signing in as admin via api. i don't want to rewrite all transactions, i want to use prepared methods of steve. from where i must start? thanks

01mk027, 2023-02-22:
under service folder ChargePointService15_Client.java file exists, if i am right, some transactions can be done by importing this file and call some methods. if i am not right, which file is proper to operate transactions? thanks

01mk027, 2023-02-22:
@juherr

goekay, 2023-02-22:
hey @01mk027, what you want to do would be part of the https://github.com/steve-community/steve/issues/1000. you want to expose the functionality as an API.

this is not as easy as you assume it to be. you can look at the rest controllers in `de.rwth.idsg.steve.web.api` as an inspiration and do something analogously.

closed 2023-02-26 by goekay