steve-community/steve #1067: What to test to qualify the status of a not listed Charging Station

whackl, 2023-02-23:
Hello ,
I've connected a Schrack Charging Station successfully to SteVe.
The Station is sending ist data through OCCP 1.5S and RFID Chips can be centrally managed through the SteVe Portal.
Anything else to test for adding the station to the supported list ?

URL : https://www.schrack.at/shop/elektromobilitaet/ladestationen/cion-ladestationen/cion-pro/i-charge-cion-pro-22kw-typ2-kabel-ethernet-ocpp-emcionp2ce.html?q=cion

The system Data as follows :

Systeminformation
Anwendungsversion
5.29.1-13419-b2601997d8
Version des Ladereglers
B94060027/19 CC613_ELPR
Linux OS version
4.14.181-gabecf650c555 #1 Fri Apr 8 23:46:21 CEST 2022 (UBI)
RCMB-Version
D0660, V2.01
Atmel Bootloader Typ
On-Die ECC (200 MHz)
Notentriegelung SW Version
D0654, V1.1.2 (Loader: D0653, V1.0.1)
EEBus-Stack-Version
KEO framework 16.2.1 (Release-16.2.1-0-g3472056a)

goekay, 2023-02-23:
hey @whackl we don't have a proper process or checklist for this wiki table. this table is maintained by the community, and therefore we rely on the goodwill of reporters.

goekay, 2023-02-26:
can we close this issue now?

whackl, 2023-02-26:
Yes, Thank you.Ive addrd it to the list, in any issues Tage place I will edit the entry.-- mit freundlichen Grüßen Ing. Dipl.Ing.(FH) Wolfgang HacklSystems Engineering S4 Informationssysteme GmbH Eduard-Bodem-Gasse 86020 InnsbruckAUSTRIA T:   +43 512 36 25 14 - 0F:   +43 512 36 25 15 - ***@***.*** W: http://www.s4.infosys.at  S4 Informationssysteme GmbH, Eduard Bodem Gasse 8, A-6020 Innsbruck; FN 106087k, Firmenbuchgericht Landesgericht Innsbruck, DVR 0770833, ATU 31848105.Diese Nachricht und allfällige angehängte Dokumente sind nur für den/die Adressaten bestimmt. Sollten Sie nicht der beabsichtigte Adressat sein, ist jede Offenlegung, Weiterleitung oder sonstige Verwendung dieser Information nicht gestattet. In diesem Fall bitten wir, den Absender zu verständigen und die Information zu vernichten. -------- Ursprüngliche Nachricht --------Von: Sevket Gökay ***@***.***> Datum: 26.02.23  17:40  (GMT+01:00) An: steve-community/steve ***@***.***> Cc: Wolfgang Hackl ***@***.***>, Mention ***@***.***> Betreff: Re: [steve-community/steve] What to test to qualify the status of a not listed Charging Station (Issue #1067) 

can we close this issue now?
—
view it on GitHub, or 
unsubscribe.

closed 2023-02-26 by goekay