steve-community/steve #1078: Steve on Docker Synology DSM 6.2.3 x64 starting error

Macrisu, 2023-02-28:
Hello
I try to install Steve 3.5.0 on a Synology DSM 6.2.3, X64 processor, via Docker

This is the error log:

```
2023-02-28 22:05:01	stderr	... 6 more
2023-02-28 22:05:01	stderr	at org.eclipse.jetty.util.component.Graceful.shutdown(Graceful.java:146)
2023-02-28 22:05:01	stderr	at java.base/java.util.stream.ReferencePipeline.toArray(ReferencePipeline.java:517)
2023-02-28 22:05:01	stderr	at java.base/java.util.stream.AbstractPipeline.evaluateToArrayNode(AbstractPipeline.java:260)
2023-02-28 22:05:01	stderr	at java.base/java.util.stream.AbstractPipeline.evaluate(AbstractPipeline.java:550)
2023-02-28 22:05:01	stderr	at java.base/java.util.stream.AbstractPipeline.wrapAndCopyInto(AbstractPipeline.java:474)
2023-02-28 22:05:01	stderr	at java.base/java.util.stream.AbstractPipeline.copyInto(AbstractPipeline.java:484)
2023-02-28 22:05:01	stderr	at java.base/java.util.ArrayList$ArrayListSpliterator.forEachRemaining(ArrayList.java:1654)
2023-02-28 22:05:01	stderr	at java.base/java.util.stream.ReferencePipeline$3$1.accept(ReferencePipeline.java:195)
2023-02-28 22:05:01	stderr	at org.eclipse.jetty.server.handler.ContextHandler.shutdown(ContextHandler.java:772)
2023-02-28 22:05:01	stderr	Caused by: java.lang.IllegalStateException: STOPPED
2023-02-28 22:05:01	stderr	at de.rwth.idsg.steve.Application.main(Application.java:68)
2023-02-28 22:05:01	stderr	at de.rwth.idsg.steve.Application.stop(Application.java:85)
2023-02-28 22:05:01	stderr	at de.rwth.idsg.steve.SteveProdStarter.stop(SteveProdStarter.java:80)
2023-02-28 22:05:01	stderr	at de.rwth.idsg.steve.JettyServer.stop(JettyServer.java:185)
2023-02-28 22:05:01	stderr	at org.eclipse.jetty.util.component.AbstractLifeCycle.stop(AbstractLifeCycle.java:132)
2023-02-28 22:05:01	stderr	at org.eclipse.jetty.server.Server.doStop(Server.java:489)
2023-02-28 22:05:01	stderr	at java.base/java.util.concurrent.CompletableFuture.get(CompletableFuture.java:2022)
2023-02-28 22:05:01	stderr	at java.base/java.util.concurrent.CompletableFuture.reportGet(CompletableFuture.java:395)
2023-02-28 22:05:01	stderr	Exception in thread "main" java.util.concurrent.ExecutionException: java.lang.IllegalStateException: STOPPED
2023-02-28 22:05:01	stdout	Please refer to the log file for details
2023-02-28 22:05:01	stdout	 FAILED!
2023-02-28 22:05:01	stdout	Starting.[INFO ] 2023-02-28 22:05:01,593 org.eclipse.jetty.server.Server (main) - jetty-10.0.11; built: 2022-06-21T21:12:44.640Z; git: d988aa016e0bb2de6fba84c1659049c72eae3e32; jvm 11.0.4+11
2023-02-28 22:05:01	stdout	Log file: Not available
2023-02-28 22:05:01	stdout	[INFO ] 2023-02-28 22:05:01,321 de.rwth.idsg.steve.Application (main) - Date/time zone of the application is set to UTC. Current date/time: 2023-02-28T22:05:01.309Z
2023-02-28 22:05:01	stdout	[INFO ] 2023-02-28 22:05:01,301 de.rwth.idsg.steve.Application (main) - Loaded the properties. Starting with the 'PROD' profile
2023-02-28 22:05:01	stdout	[INFO ] 2023-02-28 22:05:01,183 de.rwth.idsg.steve.utils.PropertiesFileLoader (main) - Hint: The Java system property 'main.properties' can be set to point to an external properties file, which will be prioritized over the bundled one
2023-02-28 22:05:00	stdout	[INFO] ------------------------------------------------------------------------
2023-02-28 22:05:00	stdout	[INFO] Finished at: 2023-02-28T22:05:00Z
2023-02-28 22:05:00	stdout	[INFO] Total time:  23.565 s
2023-02-28 22:05:00	stdout	[INFO] ------------------------------------------------------------------------
2023-02-28 22:05:00	stdout	[INFO] BUILD SUCCESS
2023-02-28 22:05:00	stdout	[INFO] ------------------------------------------------------------------------
2023-02-28 22:05:00	stdout	[INFO] Building jar: /code/target/steve.jar
2023-02-28 22:04:59	stdout	[INFO] --- maven-jar-plugin:3.3.0:jar (default-jar) @ steve ---

```

Macrisu, 2023-03-01:
I think the problem is that Synology has an older version of java: 8, for JRE

```
java -jar steve.jar
Error: A JNI error has occurred, please check your installation and try again
Exception in thread "main" java.lang.UnsupportedClassVersionError: de/rwth/idsg/steve/Application has been compiled by a more recent version of the Java Runtime (class file version 55.0), this version of the Java Runtime only recognizes class file versions up to 52.0
	at java.lang.ClassLoader.defineClass1(Native Method)
	at java.lang.ClassLoader.defineClass(ClassLoader.java:757)
	at java.security.SecureClassLoader.defineClass(SecureClassLoader.java:142)
	at java.net.URLClassLoader.defineClass(URLClassLoader.java:468)
	at java.net.URLClassLoader.access$100(URLClassLoader.java:74)
	at java.net.URLClassLoader$1.run(URLClassLoader.java:369)
	at java.net.URLClassLoader$1.run(URLClassLoader.java:363)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.net.URLClassLoader.findClass(URLClassLoader.java:362)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:419)
	at sun.misc.Launcher$AppClassLoader.loadClass(Launcher.java:352)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:352)
	at sun.launcher.LauncherHelper.checkAndLoadMain(LauncherHelper.java:601)
```

juherr, 2023-03-01:
`UnsupportedClassVersionError` means you build with a java version but run with an older java version.

If you build and run Steve via Docker, you don't care what is the installed java version. 
You should try to build and run on a classic computer before trying on a specific system like Synology.

Macrisu, 2023-03-01:
1. I use docker and see the first error.
2. I tried to run The compiled jar from docker directly and I got the second error, but I’ve installed OpenJDK11 and new error.

the problem is why is not working in docker?

emerout, 2023-03-01:
It seems that Steve can't connect to te database:
```
Caused by: com.mysql.cj.exceptions.CJCommunicationsException: Communications link failure
```

Check the configuration and the docker network where steve and the db are attached.

Dicion, 2023-03-04:
I am having similar issues with SteVe on docker.

Mine clearly connects to the DB, but ends with an error.

https://pastebin.com/8N5EjvuF

goekay, 2024-02-18:
closing due to inactivity.

closed 2024-02-18 by goekay