steve-community/steve #1094: STEVE deoloy through DOCKER unsuccessful

AndyChriss123, 2023-03-19:
Hello and thank you for looking over this issue.
When I launch STEVE using the java files everything works as expected but it seems like I cannot launch successfully using docker.
I have configured the docker main.config file as well.
I'm running on aarch64 OrangePI4.
I had to stop mysql service in order to open up the port for docker database (I don't know if this is relevant to write).
I cannot access the admin web interface on 0.0.0.0:8180/steve/manager/home

`(virtualenv) everestos@orangepi4-lts:~/STEVE/steve-steve-3.5.0$ docker compose up
[+] Running 3/3
 ⠿ Network steve-steve-350_default  Created                                                                                                  0.2s
 ⠿ Container steve-steve-350-db-1   Created                                                                                                  0.2s
 ⠿ Container steve-steve-350-app-1  Created                                                                                                  0.2s
Attaching to steve-steve-350-app-1, steve-steve-350-db-1
steve-steve-350-db-1   | 2023-03-19 13:10:54+00:00 [Note] [Entrypoint]: Entrypoint script for MariaDB Server 1:10.4.28+maria~ubu2004 started.
steve-steve-350-db-1   | 2023-03-19 13:10:54+00:00 [Note] [Entrypoint]: Switching to dedicated user 'mysql'
steve-steve-350-db-1   | 2023-03-19 13:10:54+00:00 [Note] [Entrypoint]: Entrypoint script for MariaDB Server 1:10.4.28+maria~ubu2004 started.
steve-steve-350-db-1   | 2023-03-19 13:10:55+00:00 [Note] [Entrypoint]: Initializing database files
steve-steve-350-app-1  | /bin/sh: 1: dockerize: Exec format error
steve-steve-350-app-1 exited with code 2
steve-steve-350-db-1   | 
steve-steve-350-db-1   | 
steve-steve-350-db-1   | PLEASE REMEMBER TO SET A PASSWORD FOR THE MariaDB root USER !
steve-steve-350-db-1   | To do so, start the server, then issue the following command:
steve-steve-350-db-1   | 
steve-steve-350-db-1   | '/usr/bin/mysql_secure_installation'
steve-steve-350-db-1   | 
steve-steve-350-db-1   | which will also give you the option of removing the test
steve-steve-350-db-1   | databases and anonymous user created by default.  This is
steve-steve-350-db-1   | strongly recommended for production servers.
steve-steve-350-db-1   | 
steve-steve-350-db-1   | See the MariaDB Knowledgebase at http://mariadb.com/kb
steve-steve-350-db-1   | 
steve-steve-350-db-1   | Please report any problems at http://mariadb.org/jira
steve-steve-350-db-1   | 
steve-steve-350-db-1   | The latest information about MariaDB is available at http://mariadb.org/.
steve-steve-350-db-1   | 
steve-steve-350-db-1   | Consider joining MariaDB's strong and vibrant community:
steve-steve-350-db-1   | https://mariadb.org/get-involved/
steve-steve-350-db-1   | 
steve-steve-350-db-1   | 2023-03-19 13:10:59+00:00 [Note] [Entrypoint]: Database files initialized
steve-steve-350-db-1   | 2023-03-19 13:10:59+00:00 [Note] [Entrypoint]: Starting temporary server
steve-steve-350-db-1   | 2023-03-19 13:10:59+00:00 [Note] [Entrypoint]: Waiting for server startup
steve-steve-350-db-1   | 2023-03-19 13:10:59 0 [Note] Starting MariaDB 10.4.28-MariaDB-1:10.4.28+maria~ubu2004 source revision c8f2e9a5c0ac5905f28b050b7df5a9ffd914b7e7 as process 109
steve-steve-350-db-1   | 2023-03-19 13:10:59 0 [Note] InnoDB: Using Linux native AIO
steve-steve-350-db-1   | 2023-03-19 13:10:59 0 [Note] InnoDB: Mutexes and rw_locks use GCC atomic builtins
steve-steve-350-db-1   | 2023-03-19 13:10:59 0 [Note] InnoDB: Uses event mutexes
steve-steve-350-db-1   | 2023-03-19 13:10:59 0 [Note] InnoDB: Compressed tables use zlib 1.2.11
steve-steve-350-db-1   | 2023-03-19 13:10:59 0 [Note] InnoDB: Number of pools: 1
steve-steve-350-db-1   | 2023-03-19 13:10:59 0 [Note] InnoDB: Using generic crc32 instructions
steve-steve-350-db-1   | 2023-03-19 13:10:59 0 [Note] mysqld: O_TMPFILE is not supported on /tmp (disabling future attempts)
steve-steve-350-db-1   | 2023-03-19 13:10:59 0 [Note] InnoDB: Initializing buffer pool, total size = 256M, instances = 1, chunk size = 128M
steve-steve-350-db-1   | 2023-03-19 13:10:59 0 [Note] InnoDB: Completed initialization of buffer pool
steve-steve-350-db-1   | 2023-03-19 13:10:59 0 [Note] InnoDB: If the mysqld execution user is authorized, page cleaner thread priority can be changed. See the man page of setpriority().
steve-steve-350-db-1   | 2023-03-19 13:10:59 0 [Note] InnoDB: 128 out of 128 rollback segments are active.
steve-steve-350-db-1   | 2023-03-19 13:10:59 0 [Note] InnoDB: Creating shared tablespace for temporary tables
steve-steve-350-db-1   | 2023-03-19 13:10:59 0 [Note] InnoDB: Setting file './ibtmp1' size to 12 MB. Physically writing the file full; Please wait ...
steve-steve-350-db-1   | 2023-03-19 13:10:59 0 [Note] InnoDB: File './ibtmp1' size is now 12 MB.
steve-steve-350-db-1   | 2023-03-19 13:10:59 0 [Note] InnoDB: 10.4.28 started; log sequence number 60943; transaction id 20
steve-steve-350-db-1   | 2023-03-19 13:10:59 0 [Note] Plugin 'FEEDBACK' is disabled.
steve-steve-350-db-1   | 2023-03-19 13:10:59 0 [Warning] 'user' entry 'root@b7f9a67374ce' ignored in --skip-name-resolve mode.
steve-steve-350-db-1   | 2023-03-19 13:10:59 0 [Warning] 'proxies_priv' entry '@% root@b7f9a67374ce' ignored in --skip-name-resolve mode.
steve-steve-350-db-1   | 2023-03-19 13:10:59 0 [Note] Reading of all Master_info entries succeeded
steve-steve-350-db-1   | 2023-03-19 13:10:59 0 [Note] Added new Master_info '' to hash table
steve-steve-350-db-1   | 2023-03-19 13:10:59 0 [Note] mysqld: ready for connections.
steve-steve-350-db-1   | Version: '10.4.28-MariaDB-1:10.4.28+maria~ubu2004'  socket: '/var/run/mysqld/mysqld.sock'  port: 0  mariadb.org binary distribution
steve-steve-350-db-1   | 2023-03-19 13:11:00+00:00 [Note] [Entrypoint]: Temporary server started.
steve-steve-350-db-1   | 2023-03-19 13:11:06+00:00 [Note] [Entrypoint]: GENERATED ROOT PASSWORD: yy|,K{:qug8L/s%Cg=]z1,7;pxt|/+[]
steve-steve-350-db-1   | 2023-03-19 13:11:06+00:00 [Note] [Entrypoint]: Creating database stevedb
steve-steve-350-db-1   | 2023-03-19 13:11:06+00:00 [Note] [Entrypoint]: Creating user steve
steve-steve-350-db-1   | 2023-03-19 13:11:06+00:00 [Note] [Entrypoint]: Giving user steve access to schema stevedb
[... 34 lines cut ...]
steve-steve-350-db-1   | 2023-03-19 13:11:09 0 [Note] Plugin 'FEEDBACK' is disabled.
steve-steve-350-db-1   | 2023-03-19 13:11:09 0 [Note] Server socket created on IP: '::'.
steve-steve-350-db-1   | 2023-03-19 13:11:09 0 [Note] InnoDB: Buffer pool(s) load completed at 230319 13:11:09
steve-steve-350-db-1   | 2023-03-19 13:11:09 0 [Note] Reading of all Master_info entries succeeded
steve-steve-350-db-1   | 2023-03-19 13:11:09 0 [Note] Added new Master_info '' to hash table
steve-steve-350-db-1   | 2023-03-19 13:11:09 0 [Note] mysqld: ready for connections.
steve-steve-350-db-1   | Version: '10.4.28-MariaDB-1:10.4.28+maria~ubu2004'  socket: '/var/run/mysqld/mysqld.sock'  port: 3306  mariadb.org binary distribution
steve-steve-350-db-1   | 2023-03-19 13:00:35 8 [Warning] Aborted connection 8 to db: 'unconnected' user: 'unauthenticated' host: '172.21.0.1' (This connection closed normally without authentication)
steve-steve-350-db-1   | 2023-03-19 13:00:35 9 [Warning] Aborted connection 9 to db: 'unconnected' user: 'unauthenticated' host: '172.21.0.1' (This connection closed normally without authentication)`

I'm worried about these lines in the console:

`steve-steve-350-db-1   | 2023-03-19 13:10:54+00:00 [Note] [Entrypoint]: Switching to dedicated user 'mysql'` - Shouldn't it be steve?

`steve-steve-350-app-1  | /bin/sh: 1: dockerize: Exec format error
steve-steve-350-app-1 exited with code 2`

This has been reported as soon as I tried to connect to 127.0.0.1:3306 mariadb database (which said packets out of order on the web page)

`steve-steve-350-db-1   | 2023-03-19 13:00:35 8 [Warning] Aborted connection 8 to db: 'unconnected' user: 'unauthenticated' host: '172.21.0.1' (This connection closed normally without authentication)
steve-steve-350-db-1   | 2023-03-19 13:00:35 9 [Warning] Aborted connection 9 to db: 'unconnected' user: 'unauthenticated' host: '172.21.0.1' (This connection closed normally without authentication)`

Thank you very much for your time!
If you happen to realize what is going on here and have the time to help me in any way it would be a great help to me.
All the best and good luck!

AndyChriss123, 2023-03-19:
Also this is the output from the docker network generated in case it helps somehow.

`everestos@orangepi4-lts:~$ docker network inspect steve-steve-350_default
[
    {
        "Name": "steve-steve-350_default",
        "Id": "1024e45fb518cd8cd9819752ab2214168801192b132d566a6aba5a878a79976a",
        "Created": "2023-03-19T15:10:52.73969545+02:00",
        "Scope": "local",
        "Driver": "bridge",
        "EnableIPv6": false,
        "IPAM": {
            "Driver": "default",
            "Options": null,
            "Config": [
                {
                    "Subnet": "172.25.0.0/16",
                    "Gateway": "172.25.0.1"
                }
            ]
        },
        "Internal": false,
        "Attachable": false,
        "Ingress": false,
        "ConfigFrom": {
            "Network": ""
        },
        "ConfigOnly": false,
        "Containers": {
            "b7f9a67374ce60fe4ad15ee6f159618ba98a849f98a938f5ef95045858796ad5": {
                "Name": "steve-steve-350-db-1",
                "EndpointID": "e3543f766f871638d32068075467cf5b27f0d2e468e83c889a79bca0fbe4abbc",
                "MacAddress": "02:42:ac:19:00:02",
                "IPv4Address": "172.25.0.2/16",
                "IPv6Address": ""
            }
        },
        "Options": {},
        "Labels": {
            "com.docker.compose.network": "default",
            "com.docker.compose.project": "steve-steve-350",
            "com.docker.compose.version": "2.16.0"
        }
    }
]
`

AndyChriss123, 2023-03-19:
I run on ARMBIAN OS. Do I need to make any modifications to this part of the script in Dockerfile?

`RUN wget --no-verbose https://github.com/jwilder/dockerize/releases/download/$DOCKERIZE_VERSION/dockerize-linux-amd64-$DOCKERIZE_VERSION.tar.gz \

    && tar -C /usr/local/bin -xzvf dockerize-linux-amd64-$DOCKERIZE_VERSION.tar.gz \

    && rm dockerize-linux-amd64-$DOCKERIZE_VERSION.tar.gz`

AndyChriss123, 2023-03-19:
I have found the solution. In the main repo, in the Dockerfile configuration you need to replace:

'RUN wget --no-verbose https://github.com/jwilder/dockerize/releases/download/$DOCKERIZE_VERSION/dockerize-linux-amd64-$DOCKERIZE_VERSION.tar.gz \`

`&& tar -C /usr/local/bin -xzvf dockerize-linux-amd64-$DOCKERIZE_VERSION.tar.gz \`

`&& rm dockerize-linux-amd64-$DOCKERIZE_VERSION.tar.gz`

with:
`-sfL $(curl -s https://api.github.com/repos/powerman/dockerize/releases/latest | grep -i /dockerize-$(uname -s)-$(uname -m)\" | cut -d\" -f4) | install /dev/stdin /usr/local/bin/dockerize`

in order to provide ARM64 support for steve through docker (or use docker compose commands to wait for TCP but that implies many changes).

jwilder/dockerize        - has not been updated in a very long time and could be considered discontinued.
powerman/dockerize - is a direct continuation of Dockerize which was picked up by the community.

As far as this line is concerned:
`CMD dockerize -wait tcp://mariadb:3306 -timeout 60s` 
it is a direct replacement of the original Dockerize functionality, but many other commands/features could be different. I don't know whether if it is used in other parts of the program or not so I will wait until confirmation before closing this issue. So far everything seems to be working fine.

juherr, 2023-03-20:
@AndyChriss123 Could you open a PR with your fix? But please, use `$DOCKERIZE_VERSION` instead of latest as version.

AndyChriss123, 2023-03-20:
Ok,
https://github.com/steve-community/steve/pull/1096
here is the pull request with the `$DOCKERIZE_VERSION` modification. I'm very happy if this helps anyone.
Thank you for your time and for the hard work!

closed 2023-03-21 by AndyChriss123