steve-community/steve #1100: Steve Version:	3.5.0

connermacleod69, 2023-03-24:
Hi,
I am running SteVe 3.5.0 on a Windows PC for experimentation. Connecting a Mennekes Amtron Professional+* works. But I receive an error message when I select 

http://192.168.178.40:8080/steve/manager/home/ocppJsonStatus
any of the 3 possible OCPP versions

Error:
java.lang.NullPointerException: Cannot invoke "java.lang.Integer.intValue()" because the return value of "java.util.Map.get(Object)" is null

Regards,
Serge

lategoodbye, 2023-03-24:
@connermacleod69 Please provide the complete log otherwise it's hard to figure out what's the issue here.

connermacleod69, 2023-03-28:
Hi,
I'm sorry for the response time...it works, so forget about my request. Thanks.
Regards,
Serge

closed 2023-04-15 by goekay