steve-community/steve #1109: Charger not connecting

swatilabba, 2023-03-31:
Hi Team,

I am working for an electric vehicle corporation and trying to connect Steve with Grizzl-E Smart charger but its not connecting with Steve.

**Charger details:-**
Vendor:- United Chargers
Grizzl-E  OCPP1.6J

SetUp:-
I followed Steve 3.5.0 setup on AWS line by line.
Steve application is up and running on AWS server.

Troubleshooting steps:- 
1. I turned off the firewall of my working laptop to verify if that's the issue
2. I sent a BootNotification request from Postman(on my machine) to the Steve (http://<server-host>:<server-port>/steve/services/CentralSystemService) on AWS server and I receive a heartbeat on Steve application with the request and response in the steve logs
3. But when I turn on the charger, there is nothing and I don't even see the request in the logs
4. I checked with the charger company and as per their instructions I am supposed to put
 **Enter the OCPP URL into the text box. Format the URL as follows:
• Server_DNS/Charger_ID***
5. In our case, I put either http://<server-host>:<server-port>/steve/services/CentralSystemService or http://<server-host>:<server-port>/steve/services/CentralSystemService/<ChargerSerialNumber>
6. I first hit the URL using Postman for whatever endpoint address I put in the charger to transmit data.

But the charger is still not connected! Please I need help.

Thanks!

goekay, 2023-04-09:
> Server_DNS/Charger_ID*

the fact that the instructions hint at charger id being at the end of the url makes me think that this is a WS/JSON charger. but you are using the endpoint for SOAP. pls use the url starting with `ws`

swatilabba, 2023-04-11:
I tried using WS/JSON and it didn't work too. So, I tried using virtual charger-simulator (https://github.com/vasyas/charger-simulator) to try and connect Steve. And this is what I get:- 
[image].
Please help!

goekay, 2023-04-11:
it seems like you encountered an edge case with the simulator. the action Heartbeat contains no payload. an implementation can send `null` or `{}` (empty object). steve expects `{}` whereas the simulator sends `null`.

`{}` is the suggestion by the spec:
[image]

i am not sure whether this is related to your ongoing troubles with the station though:

> I tried using WS/JSON and it didn't work too.

goekay, 2023-04-11:
> it seems like you encountered an edge case with the simulator.

and, of course, this can be seen as an edge case that steve misses as well. i guess we should support it, since it is technically acceptable even though not a suggestion.

working on a fix.

chicagz, 2023-05-03:
Has this been resolved? I am also trying to connect a Grizzl-E charger on AWS server. I tried both methods above. WS and http. I think the charger ID needs to be in the URL somewhere. I used the last 5 digits of the serial as the charger ID. I added charger ID in Steve/chargepoints

**From Grizzl-E**
Server_DNS/Charger_ID. Server DNS

**Tried the following**
SOAP: http://<your-server-ip>:<port>/steve/services/CentralSystemService
WebSocket/JSON: ws://<your-server-ip>:<port>/steve/websocket/CentralSystemService
also tried
http://<your-server-ip>:<port>/steve/services/CentralSystemService/charger_ID
ws://<your-server-ip>:<port>/steve/websocket/CentralSystemService/charger_ID 

Any thoughts on the syntax for the URL?

barsnick, 2023-05-16:
> 2. I sent a BootNotification request from Postman(on my machine) to the Steve (http://:/steve/services/CentralSystemService) on AWS server and I receive a heartbeat on Steve application with the request and response in the steve logs
> 3. But when I turn on the charger, there is nothing and I don't even see the request in the logs

Shouldn't the BootNotification be a function the charger performs?

This makes it hard to debug, if Postman (what is that?) works and the charger not, and the latter doesn't even appear in the SteVe logs.

Hints:
- Check whether SteVe's log reports a TCP connection from the charger.
- Check your charger's logs, and whether it claims to connect. (One of the Grizzl-E manuals says: "Verify that you are connected. If connection is successful the indicator light will alternate between Blue and Cyan." Is that al the information that is accessable?)
- Trace the network traffic with Wireshark. This may be tricky, as Grizzl-E is on WiFi. You might want to use tcpdump instead of Wireshark on your SteVe AWS, to see whether any packets are arriving from the charger at all.

chicagz, 2023-05-16:
I Spoke to Grizzl-E tech support. They said to remove the ws:\\ and strat with the IP address. 
(server-ip):8080/steve/services/CentralSystemService/charger_ID

I can now see a log event within steVe so I know the charger is communicating to the AWS server. But I am now getting the following error:

[ERROR] 2023-05-16 19:49:50,140 de.rwth.idsg.steve.ocpp.ws.OcppWebSocketHandshakeHandler - No protocol (OCPP version) is specified.

swatilabba, 2023-05-16:
Yes, I faced the same error. And if you look in the code because of the empty WebSocket Headers. How to fix this? I use Postman, it's a software tool that helps you to create mock requests to test your server. And in their, I get the same error. So, i don't think its the charger, it's Steve that needs some more configuration with the request.

chicagz, 2023-05-17:
Thanks for the explanation, I'll test with postman and if I find a solution, will post it here.

juherr, 2023-05-17:
The issue is clear:

The code:
```java
        List<String> requestedProtocols = new WebSocketHttpHeaders(request.getHeaders()).getSecWebSocketProtocol(); 
  
         if (CollectionUtils.isEmpty(requestedProtocols)) { 
             log.error("No protocol (OCPP version) is specified."); 
             response.setStatusCode(HttpStatus.BAD_REQUEST); 
             return false; 
         }
```

[The framework documentation](https://docs.spring.io/spring-framework/docs/current/javadoc-api/org/springframework/web/socket/WebSocketHttpHeaders.html#getSecWebSocketProtocol()):
> Returns the value of the Sec-WebSocket-Key header.

So the charger or postman must send the expected header.

juherr, 2023-05-17:
See page 7 of [the specification](https://github.com/juherr/awesome-ev-charging/blob/main/ocpp/OCPP-1.6-Documentation_2019_12/ocpp-j-1.6-specification.pdf):
[image]

Check if the charger is sending the expected header and if not, you should ask for a fix on the charger support service.

swatilabbaJPlug, 2023-05-25:
Hi guys,
Thanks for the help! Much appreciated! I learned from Grizzl-E Tech support that they are not OCPP-compliant and developing their software instead of contracting to a third party. Also, I was able to connect with the European Teltonika EV charger without any issues. But I still need to fully test it.
Hope this helps!

juherr, 2023-05-26:
@swatilabbaJPlug @swatilabba 

I wanted to add the charger model as not working in [the documentation](https://github.com/steve-community/steve/wiki/Charging-Station-Compatibility) but an entry is already there: 

> Charging Station Manufacturer | Model | Firmware | SteVe | OCPP | Status
> -- | -- | -- | -- | -- | --
> United Chargers | GRIZZL-E GRS-14-24-P | 05.41 | 3.4.5 | ocpp1.6J | working (partially tested)

Do you have the exact model and firmware of the not working charger?

No entry currently exists for the Teltonika charger.
Could you add an entry for the model you are testing too?