steve-community/steve #1114: Charge Point not showing up in Operations (steve knows CP, shows heartbeat and other data)

jb-y, 2023-04-07:
- I added Charge Point in steve (steve also picked up Charge Point under Uknown Charge Points as I turned the flag to true in main.properties)
- Charge Point station has steve's server url which is public (running on AWS LightSail)
- Steve knows the charge point as it shows model, ocpp version (1.5S) and other details
- Charge Point does not show up in operations

In summary, Steve knows the charge point station, gets all the details (ex. model, ocpp version) but doesn't show up in operations. I've attached screenshots for your reference. I am not sure what could be the issue. I can also attach logs from Steve.

[image]
[image]
[image]

goekay, 2023-04-07:
hey @jb-y since this is a charge point using soap, steve has to know the endpoint address to send messages to. as you see from your last screenshot, this is not set. therefore, steve cannot contact this station and we filter it out.

the endpoint address is set by charging station. it is one of the soap headers. apparently this is not arriving at steve.

jb-y, 2023-04-07:
thanks, that's helpful!

closed 2023-04-07 by jb-y