steve-community/steve #1119: Changing Database Port away from 3306 does not work

abc2006, 2023-04-11:
#### Checklist

- [X] I checked other issues already, but found no answer/solution
- [X] I checked the documentation and wiki, but found no answer/solution
- [X] I am running the latest version and the issue still occurs
- [X] I am sure that this issue is about SteVe (and not about the charging station software or something unrelated to SteVe)

#### Specifications
```
SteVe Version     :3.5.0
Operating system  : Ubuntu 22.04.2 LTS
JDK               : ... 
Database          : MariaDB 10.4.28-MariaDB-1:10.4.28+maria~ubu2004
```

#### Expected Behavior

docker container db and app should start up using new database port

#### Actual Behavior

docker container app does not start up

#### Steps to Reproduce the Problem

  1. wget https://github.com/steve-community/steve/archive/steve-3.5.0.tar.gz
  2. tar xvf steve-3.5.0.tar.gz
  3. cd steve-3.5.0.tar.gz
  4. change file docker-compose.yml Line 12 from 3306:3306 to 3307:3306
  5. change file src/main/resources/config/docker/main.properties:
  6. Line 10 db.port = 3306 to db.port = 3307
  7. docker-compose up

Error Message: 
app_1  | 2023/04/11 18:30:50 Connected to tcp://mariadb:3306
[...] complete log in file
[ERROR] Failed to execute goal org.flywaydb:flyway-maven-plugin:7.15.0:migrate (default) on project steve: org.flywaydb.core.internal.exception.FlywaySqlException: Unable to obtain connection from database (jdbc:mysql://mariadb:3307/stevedb?useSSL=true&serverTimezone=UTC) for user 'steve': Communications link failure

[steve2.log](https://github.com/steve-community/steve/files/11203748/steve2.log)

### What i try to do: 

i want to change the port of the mysql-container of steve to another port, so the database can coexist with the other, already on port 3306 running database. 
What am i doing wrong? 

Thank you, 
Stephan

goekay, 2023-04-11:
hey there,

the following line in your logs signal that you are using 3307 actually:

```
app_1  | [ERROR] Failed to execute goal org.flywaydb:flyway-maven-plugin:7.15.0:migrate (default) on project steve: org.flywaydb.core.internal.exception.FlywaySqlException: Unable to obtain connection from database (jdbc:mysql://mariadb:3307/stevedb?useSSL=true&serverTimezone=UTC) for user 'steve': Communications link failure
```

abc2006, 2023-04-12:
You are absolutely right. I verified that just now. 
But why does the app-container keep crashing? Or Stopping, i even cannot figure out if the container is stopping on purpose? 
How do i get steve to connect to the Database on the new port? 
If i start the decompressed files without any changes, it is running without any issues.

I can connect to the Database ( db-container keeps running) using 

```
root@dev # mysql -usteve -P3307 -pchangeme --ssl
WARNING: Forcing protocol to  TCP  due to option specification. Please explicitly state intended protocol.
Welcome to the MariaDB monitor.  Commands end with ; or \g.
Your MariaDB connection id is 12
Server version: 10.4.28-MariaDB-1:10.4.28+maria~ubu2004 mariadb.org binary distribution

Copyright (c) 2000, 2018, Oracle, MariaDB Corporation Ab and others.

Type 'help;' or '\h' for help. Type '\c' to clear the current input statement.

MariaDB [(none)]>

```
Why does docker logs give the Error
`[ERROR] Failed to execute goal org.flywaydb:flyway-maven-plugin:7.15.0:migrate (default) on project steve: org.flywaydb.core.internal.exception.FlywaySqlException: Unable to obtain connection from database (jdbc:mysql://mariadb:3307/stevedb?useSSL=true&serverTimezone=UTC) for user 'steve': Communications link failure`