steve-community/steve #1132: unable to connect to steve server

rathourcode, 2023-04-28:
steves log suggests
de.rwth.idsg.steve.ocpp.ws.OcppWebSocketHandshakeHandler - No protocol (OCPP version) is specified.

juherr, 2023-04-28:
What is the Steve issue here and what is your expectation?
If the station doesn't send the version, check there 😉

closed 2023-04-29 by rathourcode

rathourcode, 2023-05-05:
> What is the Steve issue here and what is your expectation? If the station doesn't send the version, check there 😉

hai juherr 
i am testing my steve web server , i succesfuly connected to it using postman using my local device but when i try from different device i am unable to connect to it .
i have blocked my firewalls, i have used the local ip in my websocket url what can be the possible reason for error

rathourcode, 2023-05-05:
please helppppppppp

chandan-ultraviolette, 2023-10-25:
@rathourcode  how you connected with postman ? I am trying below but facing the issue

[image]