steve-community/steve #1138: Unable to connect to steve from different computer

rathourcode, 2023-05-03:
i am testing my steve web server , i succesfuly connected to it using postman using my local device but when i try from different device i am unable to connect to it .
i have blocked my firewalls, i have used the local ip in my websocket url what can be the possible reason for error

mosea3, 2023-06-06:
which IP Address did you use to bind on?
127.0.0.1 is a loopback address which will only work on the own local device and nowhere else.
In this file you set configuration, according to the documentation
https://github.com/steve-community/steve/blob/master/src/main/resources/config/prod/main.properties#L28
Binds (serves) steve under listening address 127.0.0.1, which will not work for other computers.

You can change it to any private IP Adress in your network (eg. 192.168.1.99) or 0.0.0.0 to allow to use it from anywhere.

Further, if you mean you blocked your firewall then you need to specifically allow TCP connections in and out by port 8080.

goekay, 2023-08-20:
thanks @juherr !

closed 2023-08-20 by goekay

juherr, 2023-08-21:
You're welcome. But @mosea3 did all the job 😜

goekay, 2023-08-21:
uups, late night commenting went bad. thanks @juherr thanks @mosea3 thanks everyone! :)