steve-community/steve #1142: Build Failure

plstah, 2023-05-10:
#### Checklist

- [X ] I checked other issues already, but found no answer/solution
- [ X] I checked the documentation and wiki, but found no answer/solution
- [X ] I am running the latest version and the issue still occurs
- [ X] I am sure that this issue is about SteVe (and not about the charging station software or something unrelated to SteVe)

#### Specifications
```
SteVe Version     :3.5.0 
Operating system  : 20.04 Ubuntu
JDK               : 11
Database          : MariaDB
```

#### Expected Behavior
Hi there.
At the mvn package command i got the following error:
[INFO] BUILD FAILURE
[INFO] ------------------------------------------------------------------------
[INFO] Total time:  1.959 s
[INFO] Finished at: 2023-05-10T11:29:28Z
[INFO] ------------------------------------------------------------------------
[ERROR] Failed to execute goal org.flywaydb:flyway-maven-plugin:7.15.0:migrate (default) on project steve: org.flywaydb.core.internal.exception.FlywaySqlException: Unable to obtain connection from database (jdbc:mysql://localhost:3306/stevedb?useSSL=true&serverTimezone=UTC) for user 'steve': Access denied for user 'steve'@'localhost' (using password: YES)
[ERROR] -------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
[ERROR] SQL State  : 28000
[ERROR] Error Code : 1045
[ERROR] Message    : Access denied for user 'steve'@'localhost' (using password: YES)

I have checked the user, it does have root privileges. The stevedb exists.

Can somebody help me?

Regards.
...

#### Actual Behavior

...

#### Steps to Reproduce the Problem

  1. ...
  2. ...
  3. ...

#### Additional context

...

plstah, 2023-05-11:
> @plstah <https://github.com/plstah> Hi, this is actually not a "build
> failure". Your MariaDB server is not accessible via local TCP/SSL
> connection for the unprivileged user steve. Please fix this.

Hi Stefan,

Thank you for your reply.

In this case, what should i check? I`m not familiar with this system... :)

Regards.

plstah, 2023-05-18:
Hi there and thank you for your help!

Once i introduce that 4 rows in the config file (se attached) i can`t login
to the mariadb server.

I get the following error when i tru to restart the mariadb service: Job
for mariadb.service failed because the control process exited with error
code.
See "systemctl status mariadb.service" and "journalctl -xe" for details.

***@***.***:/home/pistah# systemctl status mariadb.service
● mariadb.service - MariaDB 10.3.38 database server
     Loaded: loaded (/lib/systemd/system/mariadb.service; enabled; vendor
preset: enabled)
     Active: failed (Result: exit-code) since Thu 2023-05-18 11:52:14 UTC;
23s ago
       Docs: man:mysqld(8)
             https://mariadb.com/kb/en/library/systemd/
    Process: 17565 ExecStartPre=/usr/bin/install -m 755 -o mysql -g root -d
/var/run/mysqld (code=exited, status=0/SUCCESS)
    Process: 17568 ExecStartPre=/bin/sh -c systemctl unset-environment
_WSREP_START_POSITION (code=exited, status=0/SUCCESS)
    Process: 17587 ExecStartPre=/bin/sh -c [ ! -e /usr/bin/galera_recovery
] && VAR= ||   VAR=`cd /usr/bin/..; /usr/bin/galera_recovery`; [ $? -eq 0 ]
  && systemctl set-environment _WSREP>
    Process: 17640 ExecStart=/usr/sbin/mysqld $MYSQLD_OPTS
$_WSREP_NEW_CLUSTER $_WSREP_START_POSITION (code=exited, status=1/FAILURE)
   Main PID: 17640 (code=exited, status=1/FAILURE)
     Status: "MariaDB server is down"

May 18 11:52:12 energy systemd[1]: Starting MariaDB 10.3.38 database
server...
May 18 11:52:14 energy systemd[1]: mariadb.service: Main process exited,
code=exited, status=1/FAILURE
May 18 11:52:14 energy systemd[1]: mariadb.service: Failed with result
'exit-code'.
May 18 11:52:14 energy systemd[1]: Failed to start MariaDB 10.3.38 database
server."

"--
-- An ExecStart= process belonging to unit mariadb.service has exited.
--
-- The process' exit code is 'exited' and its exit status is 1.
May 18 11:47:59 energy systemd[1]: mariadb.service: Failed with result
'exit-code'.
-- Subject: Unit failed
-- Defined-By: systemd
-- Support: http://www.ubuntu.com/support
--
-- The unit mariadb.service has entered the 'failed' state with result
'exit-code'.
May 18 11:47:59 energy systemd[1]: Failed to start MariaDB 10.3.38 database
server.
-- Subject: A start job for unit mariadb.service has failed
-- Defined-By: systemd
-- Support: http://www.ubuntu.com/support
--
-- A start job for unit mariadb.service has finished with a failure.
--
-- The job identifier is 6253 and the job result is failed.
May 18 11:52:12 energy systemd[1]: Starting MariaDB 10.3.38 database
server...
-- Subject: A start job for unit mariadb.service has begun execution
-- Defined-By: systemd
-- Support: http://www.ubuntu.com/support
--
-- A start job for unit mariadb.service has begun execution.
--
-- The job identifier is 6343.
May 18 11:52:14 energy systemd[1]: mariadb.service: Main process exited,
code=exited, status=1/FAILURE
-- Subject: Unit process exited
-- Defined-By: systemd
-- Support: http://www.ubuntu.com/support
--
-- An ExecStart= process belonging to unit mariadb.service has exited.
--
-- The process' exit code is 'exited' and its exit status is 1.
May 18 11:52:14 energy systemd[1]: mariadb.service: Failed with result
'exit-code'.
-- Subject: Unit failed
-- Defined-By: systemd
-- Support: http://www.ubuntu.com/support
--
-- The unit mariadb.service has entered the 'failed' state with result
'exit-code'.
May 18 11:52:14 energy systemd[1]: Failed to start MariaDB 10.3.38 database
server.
-- Subject: A start job for unit mariadb.service has failed
-- Defined-By: systemd
-- Support: http://www.ubuntu.com/support
--
-- A start job for unit mariadb.service has finished with a failure.
--
-- The job identifier is 6343 and the job result is failed.
lines 4310-4357/4357 (END)
"

goekay, 2023-05-22:
hey @plstah this issue does not seem to be related to steve. you have problems with mariadb.

closed 2023-08-20 by goekay