steve-community/steve #1147: Killed without a trace

Naheel-Azawy, 2023-05-16:
#### Checklist

- [X] I checked other issues already, but found no answer/solution
- [X] I checked the documentation and wiki, but found no answer/solution
- [ ] I am running the latest version and the issue still occurs
- [ ] I am sure that this issue is about SteVe (and not about the charging station software or something unrelated to SteVe)

#### Specifications
```
SteVe Version     : 3.4.9
Operating system  : Arch Linux, kernel 5.19.12-arch1-1
JDK               : OpenJDK 64-Bit Server VM (build 18.0.2+0, mixed mode)
Database          : mariadb  Ver 15.1 Distrib 10.9.3-MariaDB, for Linux (x86_64) using readline 5.1
```

#### Expected Behavior

The server not getting killed for no known reason

#### Actual Behavior

The server gets killed with no clear reason to be found in the logs.

I have been running steve since October last year for some experiments. Most of the experiments were done with limited amount of time and on experimental custom built setups.
Around 2 weeks ago, I connected a commercial charger to the server, Kostad CPC50, which is running 24/7. Around a week ago (rough estimate as I don't have any records for that), I noticed that steve is dead. I started it again without overthinking.
Today, I noticed the same thing happening again.

#### Steps to Reproduce the Problem

  1. Run steve for long enough
  2. Connect from a Kostad CPC50. Maybe that's unrelated, not sure
  3. Wait for week(s)
  4. Check if it gets killed

#### Additional context

I have steve running on tmux, it can be seen that it says "Killed":

```
$ java -jar target/steve.jar
Log file: /home/me/logs/steve.log
Starting............................... Done!
Hint: You can stop the application by pressing CTRL+C

Access the web interface using
- http://some.ip:8180/steve/manager/home
- http://some.other.ip:8180/steve/manager/home
- https://some.ip:8443/steve/manager/home
- https://some.other.ip:8443/steve/manager/home
SOAP endpoint for OCPP
- http://some.ip:8180/steve/services/CentralSystemService
- http://some.other.ip:8180/steve/services/CentralSystemService
- https://some.ip:8443/steve/services/CentralSystemService
- https://some.other.ip:8443/steve/services/CentralSystemService
WebSocket/JSON endpoint for OCPP
- ws://some.ip:8180/steve/websocket/CentralSystemService/(chargeBoxId)
- ws://some.other.ip:8180/steve/websocket/CentralSystemService/(chargeBoxId)
- wss://some.ip:8443/steve/websocket/CentralSystemService/(chargeBoxId)
- wss://some.other.ip:8443/steve/websocket/CentralSystemService/(chargeBoxId)

Killed
$
```

Tailing the log:

```
[INFO ] 2023-05-15 02:31:12,913 de.rwth.idsg.steve.ocpp.ws.WebSocketLogger - [chargeBoxId=SOMEID, sessionId=2761c2ed-82d5-73e8-1396-bd75d558c76b] Received: [2,"5C986B29-F13D-5F9C-001C-AC6417A65E93","Heartbeat",{}]
[INFO ] 2023-05-15 02:31:12,924 de.rwth.idsg.steve.ocpp.ws.WebSocketLogger - [chargeBoxId=SOMEID, sessionId=2761c2ed-82d5-73e8-1396-bd75d558c76b] Sending: [3,"5C986B29-F13D-5F9C-001C-AC6417A65E93",{"currentTime":"2023-05-15T02:31:12.916Z"}]
[INFO ] 2023-05-15 02:36:13,270 de.rwth.idsg.steve.ocpp.ws.WebSocketLogger - [chargeBoxId=SOMEID, sessionId=2761c2ed-82d5-73e8-1396-bd75d558c76b] Received: [2,"743D072E-413D-5F9C-001C-AC6417A65E93","Heartbeat",{}]
[INFO ] 2023-05-15 02:36:13,292 de.rwth.idsg.steve.ocpp.ws.WebSocketLogger - [chargeBoxId=SOMEID, sessionId=2761c2ed-82d5-73e8-1396-bd75d558c76b] Sending: [3,"743D072E-413D-5F9C-001C-AC6417A65E93",{"currentTime":"2023-05-15T02:36:13.272Z"}]
```

I tried grepping for "Killed" under steve's source and couldn't find anything. Is it possible that it has received a SIGKILL somehow? I'm the only person accessing this server and I doubt anyone would be interesting in putting effort to hack into my boring experiments.

My quick dirty solution for now is to run steve as follows and pray that it won't get killed while someone is charging.

```sh
while :; do date; java -jar target/steve.jar; done
```

Please excuse my outdated versions, I'm a bit worried of ruining my work. I also haven't updated steve before posting this issue because reproducing it would probably take around another week.
I'm not sure if this is a bug or if I'm doing something wrong. Any help would be appreciated.

Thanks

juherr, 2023-05-17:
It could be a memory issue.

You won't find anything in Steve logs as the process was killed outside.

What are saying system logs like `/var/log/messages`?

juherr, 2023-05-17:
Difficult to say if there is memory leak without profiling the service after many garbage collections.
Check if you can go deeper on your side.
https://www.baeldung.com/java-memory-leaks could help.

closed 2023-05-22 by Naheel-Azawy

juherr, 2023-05-22:
@Naheel-Azawy That could be nice to have a view on the requests made by the station.
Steve doesn't store them.
It will help to reproduce the exact ocpp workflow in a JMH test.