steve-community/steve #1148: About Steve 2.0.1

01mk027, 2023-05-17:
Hi dear admins and programmers;
I want to promote ocpp version of Steve to 2.0.1. As explained https://github.com/steve-community/ocpp-jaxb/blob/master/README.md, i add repository and dependency into pom.xml file. after adding dependency and repository, how i can use in software? is this mappings like a library which is included in mvn repository? and also in admin panel menu bar in operations drop down still usable ocpp versions are 1.2, 1.5 and 1.6? how i can activate 2.0.1 after adding repository and dependency?

Thanks in advance

01mk027, 2023-05-17:
must i download desired version's file and include into steve's correct path?

juherr, 2023-05-17:
Steve has already ocpp-jaxb as a dependency.
Now steve needs to support the protocol version.

Your question is duplicate of #779, #513 and #373

closed 2023-05-17 by goekay

muhmuhkuh, 2023-12-30:
Just to be sure.. is there a implementation of ocpp 2.0.1 in steve or is there just a missing „not“ in your answer? :)

goekay, 2023-12-30:
@muhmuhkuh you are right. "not" was missing in the sentence. i just edited my comment.