steve-community/steve #1149: About API and WebSocket

01mk027, 2023-05-18:
Hi admins and programmers;
For a week (maybe much more than i think), i'm trying to fire Steve's websocket mechanism by sending request to API of Steve. I want to send message with Steve's native Websocket mechanism. Because, a charge point listen only one address, that is like ws://127.0.0.1:8080/steve/websocket/CentralSystemService/WS002 and i need to send and receive messages by help of API. Is it possible? If it's not, how i can achieve? Any idea is pretty.

01mk027, 2023-05-18:
Actually, i want to send ReserveNow request from API to station.

juherr, 2023-05-18:
There is currently no API in Steve to interact with a station. See #1000

01mk027, 2023-05-20:
> There is currently no API in Steve to interact with a station. See #1000

thanks juherr

closed 2023-05-20 by goekay