steve-community/steve #1163: Steve's process will be killed at some point.

Masahirotomoda, 2023-05-31:
I installed Steve on GCP Debian following the community Configuration and Installation

The startup command is started with
>java -jar target/steve.jar &

Access is available for a few hours, but at some point access becomes inaccessible.
The life or death of the process was checked with the following commands

>ps -ef |grep java
masahir+  164702  164666  0 00:59 pts/0    00:00:00 grep java

then I've activated it again.
>java -jar target/steve.jar &
…
Hint: You can stop the application by pressing CTRL+C

Access the web interface using
- http://10.0.0.4:8080/steve/manager/home
SOAP endpoint for OCPP
- http://10.0.0.4:8080/steve/services/CentralSystemService
WebSocket/JSON endpoint for OCPP
- ws://10.0.0.4:8080/steve/websocket/CentralSystemService/(chargeBoxId)

>ps -ef |grep java
masahir+  164710  164666 68 01:01 pts/0    00:00:25 java -jar target/steve.jar
masahir+  164744  164666  0 01:02 pts/0    00:00:00 greI can a

I can access steve home.

Why is the process dying?

juherr, 2023-05-31:
What are the logs (system and steve's) saying?

Masahirotomoda, 2023-05-31:
Thx for reply Julien

Not sure if this is relevant as the time at which the process dies is 
unknown, but there is an error which indicates

[ERROR] 2023-05-22 04:37:13,045 
de.rwth.idsg.steve.ocpp.ws.WebSocketLogger - 
[chargeBoxId=AWSJ600001W1P0D2226A018A0, 
sessionId=d9461bc8-56d6-90a8-10b7-131b7c09f3a0] Transport error
java.nio.channels.ClosedChannelException: null
     at 
org.eclipse.jetty.websocket.core.internal.WebSocketSessionState.onEof(WebSocketSessionState.java:169) 
~[websocket-core-common-10.0.11.jar:10.0.11]
     at 
org.eclipse.jetty.websocket.core.internal.WebSocketCoreSession.onEof(WebSocketCoreSession.java:253) 
~[websocket-core-common-10.0.11.jar:10.0.11]
     at 
org.eclipse.jetty.websocket.core.internal.WebSocketConnection.fillAndParse(WebSocketConnection.java:482) 
~[websocket-core-common-10.0.11.jar:10.0.11]
     at 
org.eclipse.jetty.websocket.core.internal.WebSocketConnection.onFillable(WebSocketConnection.java:340) 
~[websocket-core-common-10.0.11.jar:10.0.11]
     at 
org.eclipse.jetty.io.AbstractConnection$ReadCallback.succeeded(AbstractConnection.java:319) 
~[jetty-io-10.0.11.jar:10.0.11]
     at 
org.eclipse.jetty.io.FillInterest.fillable(FillInterest.java:100) 
~[jetty-io-10.0.11.jar:10.0.11]
     at 
org.eclipse.jetty.io.SelectableChannelEndPoint$1.run(SelectableChannelEndPoint.java:53) 
~[jetty-io-10.0.11.jar:10.0.11]
     at 
org.eclipse.jetty.util.thread.strategy.AdaptiveExecutionStrategy.runTask(AdaptiveExecutionStrategy.java:412) 
~[jetty-util-10.0.11.jar:10.0.11]
     at 
org.eclipse.jetty.util.thread.strategy.AdaptiveExecutionStrategy.consumeTask(AdaptiveExecutionStrategy.java:381) 
~[jetty-util-10.0.11.jar:10.0.11]
     at 
org.eclipse.jetty.util.thread.strategy.AdaptiveExecutionStrategy.tryProduce(AdaptiveExecutionStrategy.java:268) 
~[jetty-util-10.0.11.jar:10.0.11]
     at 
org.eclipse.jetty.util.thread.strategy.AdaptiveExecutionStrategy.lambda$new$0(AdaptiveExecutionStrategy.java:138) 
~[jetty-util-10.0.11.jar:10.0.11]
     at 
org.eclipse.jetty.util.thread.ReservedThreadExecutor$ReservedThread.run(ReservedThreadExecutor.java:407) 
~[jetty-util-10.0.11.jar:10.0.11]
     at 
org.eclipse.jetty.util.thread.QueuedThreadPool.runJob(QueuedThreadPool.java:894) 
~[jetty-util-10.0.11.jar:10.0.11]
     at 
org.eclipse.jetty.util.thread.QueuedThreadPool$Runner.run(QueuedThreadPool.java:1038) 
~[jetty-util-10.0.11.jar:10.0.11]
     at java.lang.Thread.run(Thread.java:829) ~[?:?]
[WARN ] 2023-05-22 04:37:13,046 
de.rwth.idsg.steve.ocpp.ws.WebSocketLogger - 
[chargeBoxId=AWSJ600001W1P0D2226A018A0, 
sessionId=d9461bc8-56d6-90a8-10b7-131b7c09f3a0] Connection is closed, 
status: CloseStatus[code=1006, reason=Session Closed]

juherr, 2023-05-31:
Check system logs. Your issue could be related to #1147 / #509

Masahirotomoda, 2023-06-04:
I search kern.log

May 26 11:39:01 ocpp-ap01 kernel: [2336070.094330] Tasks state (memory values in pages):
May 26 11:39:01 ocpp-ap01 kernel: [2336070.099515] [  pid  ]   uid  tgid total_vm      rss pgtables_bytes swapents oom_score_adj name
May 26 11:39:01 ocpp-ap01 kernel: [2336070.108618] [    316]     0   316     2046     1109    53248        0             0 haveged
May 26 11:39:01 ocpp-ap01 kernel: [2336070.117280] [    336]   103   336     2025      181    53248        0          -900 dbus-daemon
May 26 11:39:01 ocpp-ap01 kernel: [2336070.126289] [    340]     0   340    55234      421    77824        0             0 rsyslogd
May 26 11:39:01 ocpp-ap01 kernel: [2336070.135040] [    344]     0   344    24973      829    73728        0             0 dhclient
May 26 11:39:01 ocpp-ap01 kernel: [2336070.143806] [    411]     0   411   180944     3390   118784        0          -999 google_guest_ag
May 26 11:39:01 ocpp-ap01 kernel: [2336070.153611] [    412]     0   412   182581     3989   143360        0             0 google_osconfig
May 26 11:39:01 ocpp-ap01 kernel: [2336070.163106] [    429]   105   429     4763      175    61440        0             0 chronyd
May 26 11:39:01 ocpp-ap01 kernel: [2336070.171943] [    430]   105   430     2682      138    61440        0             0 chronyd
May 26 11:39:01 ocpp-ap01 kernel: [2336070.180652] [    472]     0   472     6649     1786    86016        0             0 unattended-upgr
May 26 11:39:01 ocpp-ap01 kernel: [2336070.190011] [    500]     0   500      719       29    45056        0             0 agetty
May 26 11:39:01 ocpp-ap01 kernel: [2336070.198685] [    501]     0   501     1100       30    45056        0             0 agetty
May 26 11:39:01 ocpp-ap01 kernel: [2336070.207276] [    517]     0   517     3340      244    69632        0         -1000 sshd
May 26 11:39:01 ocpp-ap01 kernel: [2336070.215712] [    533]     0   533     3447      223    69632        0             0 systemd-logind
May 26 11:39:01 ocpp-ap01 kernel: [2336070.225366] [   5467]     0  5467     1410       67    45056        0             0 cron
May 26 11:39:01 ocpp-ap01 kernel: [2336070.233770] [   9941]     0  9941    12083      280   110592        0          -250 systemd-journal
May 26 11:39:01 ocpp-ap01 kernel: [2336070.243132] [  10871]     0 10871     4809      197    65536        0         -1000 systemd-udevd
May 26 11:39:01 ocpp-ap01 kernel: [2336070.252706] [ 103090]  1000 103090     3764      238    65536        0             0 systemd
May 26 11:39:01 ocpp-ap01 kernel: [2336070.261627] [ 103091]  1000 103091    41917      687    94208        0             0 (sd-pam)
May 26 11:39:01 ocpp-ap01 kernel: [2336070.270463] [ 103114]  1000 103114   982456   452302  4009984        0             0 java
May 26 11:39:01 ocpp-ap01 kernel: [2336070.278945] [ 145335]     0 145335   180235       53    94208        0             0 gce_workload_ce
May 26 11:39:01 ocpp-ap01 kernel: [2336070.288510] oom-kill:constraint=CONSTRAINT_NONE,nodemask=(null),cpuset=/,mems_allowed=0,global_oom,task_memcg=/user.slice/user-1000.slice/session-553.scope,task=java,pid=103114,uid=1000

May 26 11:39:01 ocpp-ap01 kernel: [2336070.305478] Out of memory: Killed process 103114 (java) total-vm:3929824kB, anon-rss:1809208kB, file-rss:0kB, shmem-rss:0kB, UID:1000 pgtables:3916kB oom_score_adj:0

The process killed cause out of memory,

while :; do date; java -jar target/steve.jar; done

Would this workaround work?
Also, does charging stop if the vehicle is charging when the process dies?

goekay, 2023-06-04:
> Also, does charging stop if the vehicle is charging when the process dies?

no. charging stations are autonomous in that regard. they are just notifying steve start and stop.

which steve version are you using? there has been a memory problem which was fixed recently. see #846 and #1058

Masahirotomoda, 2023-06-04:
Hello GoKay

Thx for this ocpp software.

Im using 3.5.0

Im understand this version has memory problem.

and Is this version not supported api?

Should I downgrade my software version?

On 2023/06/04 15:46, Sevket Gökay wrote:
> which steve version are you using? there has been a memory problem 
> which was fixed recently.

goekay, 2023-06-04:
i actually just [made a release](https://github.com/steve-community/steve/releases/tag/steve-3.6.0) which includes the fix and the APIs for 3rd parties

mtomoda, 2023-06-09:
It has been 3 days since we installed the new version and the process no longer dies.
Thx goekay.

I want to find API usage.
Example start remote transaction.
Does anyone know of a URL with more information?

juherr, 2023-06-09:
@mtomoda see #1000

goekay, 2023-06-10:
> I want to find API usage.
> Example start remote transaction.
> Does anyone know of a URL with more information?

@mtomoda the capabilities of the API are limited. you cannot start remote transaction via API call. there is no endpoint for this.

mtomoda, 2023-06-10:
Thank you reply goekay.

I see
I want to start charging by smartphone.
Smartphone --- CentralServer(Steve) --- ChargePoint

I believe it is possible to run ChargePoint through Steve from my smartphone via SOAP or WebSocket rather than API.
Is there a website that shows how to make this connection?

juherr, 2023-06-10:
@mtomoda It will possible when someone will implement it. Maybe a fork already do it but it should be considered like another project.

mtomoda, 2023-06-11:
Thx for reply @juherr
Ill try it.

goekay, 2023-07-25:
closing this issue since it seems to be resolved.

closed 2023-07-25 by goekay