steve-community/steve #1169: SteVe trying to handle the request instead of forwarding it to the charger

justarandomnerd, 2023-06-09:
I am trying to implement a dynamic smart charging functionality on an EV charger using python scripts and using SteVe just as a backup back-end. I am running scripts on the same computer as SteVe is running on, but instead of forwarding the request to the charger, I believe SteVe is trying to handle it itself.
For example, If I make a GetConfiguration request through the script, I send the message to the endpoint of charger specified which is 
ws://<IP_address>:<port>/steve/websocket/CentralSystemService/<charger_name>
[2,"aarpkeytest","GetConfiguration",{"key":[]}]
Received message: [4,"aarpkeytest","NotImplemented","The action 'GetConfiguration' you are looking for is not found",{}]
whereas I get the actual expected response if I make the request through SteVe's web UI, which essentially send the same message, but I get the expected response.
Any insights on what I am doing wrong or how to accomplish this?

juherr, 2023-06-10:
Steve is not an ocpp gateway that forwards messages but a central point where all stations are sending messages.

You are trying to connect your python script on the ocpp interface dedicated to stations, right? 
But a station cannot interract with another station like that.

What you need is a way to tell Steve what you want and it will contact the station itself.

Currently there is few feature and you should have a look at #1000

justarandomnerd, 2023-06-12:
Hi, Thank you for your response, I looked into the thread and also went through the api-documentation. I have a brief understanding not on how it works, but unfortunately at the moment SteVe does not seem to support the functionality I am looking for and as it is extremely user specific, I highly doubt it would be worth a feature request. Thankfully, it seems that there seems to be a base ready where in I can add the functionality I need, I just need to figure out how to add the new curl commands. I'm closing the issue but feel free to add additional comments on where I could find resources to add new curl commands.

justarandomnerd, 2023-06-12:
Hi, Thank you for your response, I looked into the thread and also went through the api-documentation. I have a brief understanding not on how it works, but unfortunately at the moment SteVe does not seem to support the functionality I am looking for and as it is extremely user specific, I highly doubt it would be worth a feature request. Thankfully, it seems that there seems to be a base ready where in I can add the functionality I need, I just need to figure out how to add the new curl commands. I'm closing the issue but feel free to add additional comments on where I could find resources to add new curl comman

closed 2023-06-12 by justarandomnerd