steve-community/steve #1174: Charging Profile Rejected by EV Charger

justarandomnerd, 2023-06-12:
I'm working with an actual EV charger and I'm trying to set a charging profile.
I am adding a charging profile through Data Management > Charging Profiles
and the trying to upload it to the charger through Operations > OCPP v1.6 > Set Charging Profile.
But when I perform the action, I am getting a response "Rejected"
Any idea on what would be the issue and how to resolve it?

justarandomnerd, 2023-06-12:
@juherr 
[INFO ] 2023-06-12 20:13:14,453 de.rwth.idsg.steve.ocpp.ws.WebSocketLogger (SteVe-Executor-0) - [chargeBoxId=<chargebox_id>, sessionId=7a755952-a574-bb89-da8d-5a018c39fa60] Sending: [2,"ae419f44-690a-468b-9e10-4d408e6e2802","SetChargingProfile",{"connectorId":0,"csChargingProfiles":{"chargingProfileId":1,"stackLevel":1,"chargingProfilePurpose":"ChargePointMaxProfile","chargingProfileKind":"Absolute","validFrom":"2023-01-01T00:00:00.000Z","validTo":"2023-12-15T00:00:00.000Z","chargingSchedule":{"duration":120,"chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":120,"limit":10,"numberPhases":3}],"minChargingRate":8}}}]
[INFO ] 2023-06-12 20:13:14,743 de.rwth.idsg.steve.ocpp.ws.WebSocketLogger (qtp65080774-24) - [chargeBoxId=<chargebox_id>, sessionId=7a755952-a574-bb89-da8d-5a018c39fa60] Received: [3, "ae419f44-690a-468b-9e10-4d408e6e2802", {"status": "Rejected"}]

This is the SteVe log.

justarandomnerd, 2023-06-12:
Also, is there any method through which I can send customized commands to SteVe?
As in any where I could type out a command instead of clicking on the UI? For example I manually type
[2,"auth_key", "GetConfiguration",{}]
and it gives the same response of what would happen if I clicked Operations > OCPP v1.6 > Get Configuration > Perform

juherr, 2023-06-13:
> Also, is there any method through which I can send customized commands to SteVe?
> As in any where I could type out a command instead of clicking on the UI? For example I manually type
> [2,"auth_key", [...]

Not yet. See #1000

closed 2023-06-26 by justarandomnerd