steve-community/steve #1177: Procedure to set charging limit to connectorid

sushant9930, 2023-06-23:
I had lot of try for setCharging profile to connectorId but, I didn't understand procedure of set charging profile to connectorId. 
[image]
[image]
i get this kind of response return to me from steve can any one explain me step by step procedure to set limit to running connectorId.
and internally its show me ( 'child "chargingSchedule" fails because [child "chargingSchedulePeriod" fails because ["chargingSchedulePeriod" at position 0 fails because [child "limit" fails because ["limit" must be a multiple of 0.1]]]]')
 this error and my charging profile is like ([2,"79b70326-6153-4963-9708-045d9a8f0d1c","SetChargingProfile",{"connectorId":1,"csChargingProfiles":{"chargingProfileId":1,"stackLevel":1,"chargingProfilePurpose":"TxProfile","chargingProfileKind":"Recurring","recurrencyKind":"Daily","chargingSchedule":{"duration":1000,"chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":7600,"limit":124,"numberPhases":1}],"minChargingRate":10}}}]
)

fnkbsi, 2023-06-23:
Steve does not support TxProfiles (yet?). You can use ChargePointMaxProfile and TxDefaultProfile, how they work is described in the OCPP documentation.

Please read also #826 and #367

sushant9930, 2023-06-23:
so you means TxProfiles is also not support in "smart charging"?

fnkbsi, 2023-06-23:
I think so. But if I remember correctly, the TxProfile only needs additionally the TransactionID and Starttime. In one of the forks I saw a solution for testing purposes, sorry can't remember in which one

sushant9930, 2023-06-23:
Sorry I want to correct some information. I was get  ( 'child "chargingSchedule" fails because [child "chargingSchedulePeriod" fails because ["chargingSchedulePeriod" at position 0 fails because [child "limit" fails because ["limit" must be a multiple of 0.1]]]]') this error from client side actually. when i was pass TxProfile charging profile and this message is like limit must be a multiple of 0.1 but i can't understand why this error occured and this error comes with TxDefaultProfile charging purpose also

fnkbsi, 2023-06-23:
Not sure, but is your limit in the profile 124 or 124.0 also the minChargingRate should be 10.0

sushant9930, 2023-06-23:
[image]
this error showed me at client server when i am trying to setchargingprofile and could please you give me documentation which is best for understand proper procedure of setchanging to particular connectorid else any video documentation

closed 2025-09-16 by goekay