steve-community/steve #1178: I have the following error

andreschinchilla81, 2023-06-23:
I have the following error when I want to consult all the transactions:

steve/manager/transactions/query raised following exception.
java.lang.ClassCastException: class java.lang.Double cannot be cast to class java.lang.String (java.lang.Double and java.lang.String are in module java.base of loader 'bootstrap')

Can you help me please?

juherr, 2023-06-24:
What is your steve version? What are saying the logs?

andreschinchilla81, 2023-06-26:
Thanks!

My steve version is 3.6.0

This is the log:
[ERROR] 2023-06-26 08:35:43,154 de.rwth.idsg.steve.web.GlobalControllerAdvice - Request: http://172.16.1.4:8081/steve/manager/transactions/query raised following exception.
java.lang.ClassCastException: class java.lang.Double cannot be cast to class java.lang.String (java.lang.Double and java.lang.String are in module java.base of loader 'bootstrap')
	at de.rwth.idsg.steve.repository.impl.TransactionRepositoryImpl$TransactionMapper.map(TransactionRepositoryImpl.java:361) ~[steve.jar:?]
	at de.rwth.idsg.steve.repository.impl.TransactionRepositoryImpl$TransactionMapper.map(TransactionRepositoryImpl.java:348) ~[steve.jar:?]
	at org.jooq.impl.ResultImpl.lambda$map$0(ResultImpl.java:936) ~[jooq-3.16.7.jar:?]
	at org.jooq.impl.Tools.map(Tools.java:2134) ~[jooq-3.16.7.jar:?]
	at org.jooq.impl.ResultImpl.map(ResultImpl.java:936) ~[jooq-3.16.7.jar:?]
	at de.rwth.idsg.steve.repository.impl.TransactionRepositoryImpl.getTransactions(TransactionRepositoryImpl.java:71) ~[steve.jar:?]
	at de.rwth.idsg.steve.web.controller.TransactionsReservationsController.getTransactionsQuery(TransactionsReservationsController.java:117) ~[steve.jar:?]
	at jdk.internal.reflect.DirectMethodHandleAccessor.invoke(DirectMethodHandleAccessor.java:104) ~[?:?]
	at java.lang.reflect.Method.invoke(Method.java:578) ~[?:?]
	at org.springframework.web.method.support.InvocableHandlerMethod.doInvoke(InvocableHandlerMethod.java:205) ~[spring-web-5.3.23.jar:5.3.23]
	at org.springframework.web.method.support.InvocableHandlerMethod.invokeForRequest(InvocableHandlerMethod.java:150) ~[spring-web-5.3.23.jar:5.3.23]
	at org.springframework.web.servlet.mvc.method.annotation.ServletInvocableHandlerMethod.invokeAndHandle(ServletInvocableHandlerMethod.java:117) ~[spring-webmvc-5.3.23.jar:5.3.23]
	at org.springframework.web.servlet.mvc.method.annotation.RequestMappingHandlerAdapter.invokeHandlerMethod(RequestMappingHandlerAdapter.java:895) ~[spring-webmvc-5.3.23.jar:5.3.23]
	at org.springframework.web.servlet.mvc.method.annotation.RequestMappingHandlerAdapter.handleInternal(RequestMappingHandlerAdapter.java:808) ~[spring-webmvc-5.3.23.jar:5.3.23]
	at org.springframework.web.servlet.mvc.method.AbstractHandlerMethodAdapter.handle(AbstractHandlerMethodAdapter.java:87) ~[spring-webmvc-5.3.23.jar:5.3.23]
	at org.springframework.web.servlet.DispatcherServlet.doDispatch(DispatcherServlet.java:1071) ~[spring-webmvc-5.3.23.jar:5.3.23]
	at org.springframework.web.servlet.DispatcherServlet.doService(DispatcherServlet.java:964) ~[spring-webmvc-5.3.23.jar:5.3.23]
	at org.springframework.web.servlet.FrameworkServlet.processRequest(FrameworkServlet.java:1006) ~[spring-webmvc-5.3.23.jar:5.3.23]
	at org.springframework.web.servlet.FrameworkServlet.doGet(FrameworkServlet.java:898) ~[spring-webmvc-5.3.23.jar:5.3.23]
	at javax.servlet.http.HttpServlet.service(HttpServlet.java:503) ~[jetty-servlet-api-4.0.6.jar:?]
	at org.springframework.web.servlet.FrameworkServlet.service(FrameworkServlet.java:883) ~[spring-webmvc-5.3.23.jar:5.3.23]
	at javax.servlet.http.HttpServlet.service(HttpServlet.java:590) ~[jetty-servlet-api-4.0.6.jar:?]
	at org.eclipse.jetty.servlet.ServletHolder.handle(ServletHolder.java:764) ~[jetty-servlet-10.0.13.jar:10.0.13]
	at org.eclipse.jetty.servlet.ServletHandler$ChainEnd.doFilter(ServletHandler.java:1665) ~[jetty-servlet-10.0.13.jar:10.0.13]
	at org.springframework.security.web.FilterChainProxy$VirtualFilterChain.doFilter(FilterChainProxy.java:337) ~[spring-security-web-5.7.5.jar:5.7.5]
	at org.springframework.security.web.access.ExceptionTranslationFilter.doFilter(ExceptionTranslationFilter.java:122) ~[spring-security-web-5.7.5.jar:5.7.5]
	at org.springframework.security.web.access.ExceptionTranslationFilter.doFilter(ExceptionTranslationFilter.java:116) ~[spring-security-web-5.7.5.jar:5.7.5]
	at org.springframework.security.web.FilterChainProxy$VirtualFilterChain.doFilter(FilterChainProxy.java:346) ~[spring-security-web-5.7.5.jar:5.7.5]
	at org.springframework.security.web.session.SessionManagementFilter.doFilter(SessionManagementFilter.java:126) ~[spring-security-web-5.7.5.jar:5.7.5]
	at org.springframework.security.web.session.SessionManagementFilter.doFilter(SessionManagementFilter.java:81) ~[spring-security-web-5.7.5.jar:5.7.5]
	at org.springframework.security.web.FilterChainProxy$VirtualFilterChain.doFilter(FilterChainProxy.java:346) ~[spring-security-web-5.7.5.jar:5.7.5]
	at org.springframework.security.web.authentication.AnonymousAuthenticationFilter.doFilter(AnonymousAuthenticationFilter.java:109) ~[spring-security-web-5.7.5.jar:5.7.5]
	at org.springframework.security.web.FilterChainProxy$VirtualFilterChain.doFilter(FilterChainProxy.java:346) ~[spring-security-web-5.7.5.jar:5.7.5]
	at org.springframework.security.web.servletapi.SecurityContextHolderAwareRequestFilter.doFilter(SecurityContextHolderAwareRequestFilter.java:149) ~[spring-security-web-5.7.5.jar:5.7.5]
	at org.springframework.security.web.FilterChainProxy$VirtualFilterChain.doFilter(FilterChainProxy.java:346) ~[spring-security-web-5.7.5.jar:5.7.5]
	at org.springframework.security.web.savedrequest.RequestCacheAwareFilter.doFilter(RequestCacheAwareFilter.java:63) ~[spring-security-web-5.7.5.jar:5.7.5]
	at org.springframework.security.web.FilterChainProxy$VirtualFilterChain.doFilter(FilterChainProxy.java:346) ~[spring-security-web-5.7.5.jar:5.7.5]
[... cut ...]
	at org.eclipse.jetty.util.thread.strategy.AdaptiveExecutionStrategy.lambda$new$0(AdaptiveExecutionStrategy.java:140) ~[jetty-util-10.0.13.jar:10.0.13]
	at org.eclipse.jetty.util.thread.ReservedThreadExecutor$ReservedThread.run(ReservedThreadExecutor.java:411) ~[jetty-util-10.0.13.jar:10.0.13]
	at org.eclipse.jetty.util.thread.QueuedThreadPool.runJob(QueuedThreadPool.java:934) ~[jetty-util-10.0.13.jar:10.0.13]
	at org.eclipse.jetty.util.thread.QueuedThreadPool$Runner.run(QueuedThreadPool.java:1078) ~[jetty-util-10.0.13.jar:10.0.13]
	at java.lang.Thread.run(Thread.java:1589) [?:?]

andreschinchilla81, 2023-06-26:
> What is your steve version? What are saying the logs?

Thanks!

My steve version is 3.6.0

This is the log:
[ERROR] 2023-06-26 08:35:43,154 de.rwth.idsg.steve.web.GlobalControllerAdvice - Request: http://172.16.1.4:8081/steve/manager/transactions/query raised following exception.
java.lang.ClassCastException: class java.lang.Double cannot be cast to class java.lang.String (java.lang.Double and java.lang.String are in module java.base of loader 'bootstrap')
at de.rwth.idsg.steve.repository.impl.TransactionRepositoryImpl$TransactionMapper.map(TransactionRepositoryImpl.java:361) ~[steve.jar:?]
at de.rwth.idsg.steve.repository.impl.TransactionRepositoryImpl$TransactionMapper.map(TransactionRepositoryImpl.java:348) ~[steve.jar:?]
at org.jooq.impl.ResultImpl.lambda$map$0(ResultImpl.java:936) ~[jooq-3.16.7.jar:?]
at org.jooq.impl.Tools.map(Tools.java:2134) ~[jooq-3.16.7.jar:?]
at org.jooq.impl.ResultImpl.map(ResultImpl.java:936) ~[jooq-3.16.7.jar:?]
at de.rwth.idsg.steve.repository.impl.TransactionRepositoryImpl.getTransactions(TransactionRepositoryImpl.java:71) ~[steve.jar:?]
at de.rwth.idsg.steve.web.controller.TransactionsReservationsController.getTransactionsQuery(TransactionsReservationsController.java:117) ~[steve.jar:?]
at jdk.internal.reflect.DirectMethodHandleAccessor.invoke(DirectMethodHandleAccessor.java:104) ~[?:?]
at java.lang.reflect.Method.invoke(Method.java:578) ~[?:?]
at org.springframework.web.method.support.InvocableHandlerMethod.doInvoke(InvocableHandlerMethod.java:205) ~[spring-web-5.3.23.jar:5.3.23]
at org.springframework.web.method.support.InvocableHandlerMethod.invokeForRequest(InvocableHandlerMethod.java:150) ~[spring-web-5.3.23.jar:5.3.23]
at org.springframework.web.servlet.mvc.method.annotation.ServletInvocableHandlerMethod.invokeAndHandle(ServletInvocableHandlerMethod.java:117) ~[spring-webmvc-5.3.23.jar:5.3.23]
at org.springframework.web.servlet.mvc.method.annotation.RequestMappingHandlerAdapter.invokeHandlerMethod(RequestMappingHandlerAdapter.java:895) ~[spring-webmvc-5.3.23.jar:5.3.23]
at org.springframework.web.servlet.mvc.method.annotation.RequestMappingHandlerAdapter.handleInternal(RequestMappingHandlerAdapter.java:808) ~[spring-webmvc-5.3.23.jar:5.3.23]
at org.springframework.web.servlet.mvc.method.AbstractHandlerMethodAdapter.handle(AbstractHandlerMethodAdapter.java:87) ~[spring-webmvc-5.3.23.jar:5.3.23]
at org.springframework.web.servlet.DispatcherServlet.doDispatch(DispatcherServlet.java:1071) ~[spring-webmvc-5.3.23.jar:5.3.23]
at org.springframework.web.servlet.DispatcherServlet.doService(DispatcherServlet.java:964) ~[spring-webmvc-5.3.23.jar:5.3.23]
at org.springframework.web.servlet.FrameworkServlet.processRequest(FrameworkServlet.java:1006) ~[spring-webmvc-5.3.23.jar:5.3.23]
at org.springframework.web.servlet.FrameworkServlet.doGet(FrameworkServlet.java:898) ~[spring-webmvc-5.3.23.jar:5.3.23]
at javax.servlet.http.HttpServlet.service(HttpServlet.java:503) ~[jetty-servlet-api-4.0.6.jar:?]
at org.springframework.web.servlet.FrameworkServlet.service(FrameworkServlet.java:883) ~[spring-webmvc-5.3.23.jar:5.3.23]
at javax.servlet.http.HttpServlet.service(HttpServlet.java:590) ~[jetty-servlet-api-4.0.6.jar:?]
at org.eclipse.jetty.servlet.ServletHolder.handle(ServletHolder.java:764) ~[jetty-servlet-10.0.13.jar:10.0.13]
at org.eclipse.jetty.servlet.ServletHandler$ChainEnd.doFilter(ServletHandler.java:1665) ~[jetty-servlet-10.0.13.jar:10.0.13]
at org.springframework.security.web.FilterChainProxy$VirtualFilterChain.doFilter(FilterChainProxy.java:337) ~[spring-security-web-5.7.5.jar:5.7.5]
at org.springframework.security.web.access.ExceptionTranslationFilter.doFilter(ExceptionTranslationFilter.java:122) ~[spring-security-web-5.7.5.jar:5.7.5]
at org.springframework.security.web.access.ExceptionTranslationFilter.doFilter(ExceptionTranslationFilter.java:116) ~[spring-security-web-5.7.5.jar:5.7.5]
at org.springframework.security.web.FilterChainProxy$VirtualFilterChain.doFilter(FilterChainProxy.java:346) ~[spring-security-web-5.7.5.jar:5.7.5]
at org.springframework.security.web.session.SessionManagementFilter.doFilter(SessionManagementFilter.java:126) ~[spring-security-web-5.7.5.jar:5.7.5]
at org.springframework.security.web.session.SessionManagementFilter.doFilter(SessionManagementFilter.java:81) ~[spring-security-web-5.7.5.jar:5.7.5]
at org.springframework.security.web.FilterChainProxy$VirtualFilterChain.doFilter(FilterChainProxy.java:346) ~[spring-security-web-5.7.5.jar:5.7.5]
at org.springframework.security.web.authentication.AnonymousAuthenticationFilter.doFilter(AnonymousAuthenticationFilter.java:109) ~[spring-security-web-5.7.5.jar:5.7.5]
at org.springframework.security.web.FilterChainProxy$VirtualFilterChain.doFilter(FilterChainProxy.java:346) ~[spring-security-web-5.7.5.jar:5.7.5]
at org.springframework.security.web.servletapi.SecurityContextHolderAwareRequestFilter.doFilter(SecurityContextHolderAwareRequestFilter.java:149) ~[spring-security-web-5.7.5.jar:5.7.5]
at org.springframework.security.web.FilterChainProxy$VirtualFilterChain.doFilter(FilterChainProxy.java:346) ~[spring-security-web-5.7.5.jar:5.7.5]
at org.springframework.security.web.savedrequest.RequestCacheAwareFilter.doFilter(RequestCacheAwareFilter.java:63) ~[spring-security-web-5.7.5.jar:5.7.5]
at org.springframework.security.web.FilterChainProxy$VirtualFilterChain.doFilter(FilterChainProxy.java:346) ~[spring-security-web-5.7.5.jar:5.7.5]
[... cut ...]
at org.eclipse.jetty.util.thread.strategy.AdaptiveExecutionStrategy.lambda$new$0(AdaptiveExecutionStrategy.java:140) ~[jetty-util-10.0.13.jar:10.0.13]
at org.eclipse.jetty.util.thread.ReservedThreadExecutor$ReservedThread.run(ReservedThreadExecutor.java:411) ~[jetty-util-10.0.13.jar:10.0.13]
at org.eclipse.jetty.util.thread.QueuedThreadPool.runJob(QueuedThreadPool.java:934) ~[jetty-util-10.0.13.jar:10.0.13]
at org.eclipse.jetty.util.thread.QueuedThreadPool$Runner.run(QueuedThreadPool.java:1078) ~[jetty-util-10.0.13.jar:10.0.13]
at java.lang.Thread.run(Thread.java:1589) [?:?]

juherr, 2023-06-28:
The issue comes from https://github.com/steve-community/steve/blob/master/src/main/java/de/rwth/idsg/steve/repository/impl/TransactionRepositoryImpl.java#L361

For an unknown reason, `r.value8()` returns a `Double` instead of the expected `String`: https://github.com/steve-community/steve/blob/master/src/main/java/de/rwth/idsg/steve/repository/dto/Transaction.java#L69

What is the database and its version?
Could you check the scheme of the transaction table? Is the stop_value column VARCHAR or FLOAT?
What is happening if you downgrade to the previous steve version?

andreschinchilla81, 2023-07-15:
If the database is indeed returning a Double, you can convert it to a String using the String.valueOf() method when setting the stopValue:
public Transaction map(Record12<Integer, String, Integer, String, DateTime, String, DateTime, String, String, Integer, Integer, TransactionStopEventActor> r) {
    return Transaction.builder()
                      .id(r.value1())
                      .chargeBoxId(r.value2())
                      .connectorId(r.value3())
                      .ocppIdTag(r.value4())
                      .startTimestamp(r.value5())
                      .startTimestampFormatted(DateTimeUtils.humanize(r.value5()))
                      .startValue(r.value6())
                      .stopTimestamp(r.value7())
                      .stopTimestampFormatted(DateTimeUtils.humanize(r.value7()))
                      .stopValue(String.valueOf(r.value8())) // Convert Double to String
                      .stopReason(r.value9())
                      .chargeBoxPk(r.value10())
                      .ocppTagPk(r.value11())
                      .stopEventActor(r.value12())
                      .build();

juherr, 2023-07-16:
@andreschinchilla81 Your solution won't compile because `r.value8()` is already declared to return `String`. The jooq files are generated from the database scheme and `stop_value` is supposed to be a `VARCHAR`:

```java
    public final TableField<TransactionRecord, String> STOP_VALUE = createField(DSL.name("stop_value"), SQLDataType.VARCHAR(255), this, "");
```

I think your issue comes from your specific context and you'll have to share more in order to be able to reproduce the issue.

andreschinchilla81, 2023-07-16:
@juherr it works!
[image]

> @andreschinchilla81 Your solution won't compile because `r.value8()` is already declared to return `String`. The jooq files are generated from the database scheme and `stop_value` is supposed to be [...]

closed 2023-07-16 by andreschinchilla81

juherr, 2023-07-16:
@andreschinchilla81 You had right your solution compiles because it is possible to apply `toString()` on every object.
Did you find the reason of your issue?

andreschinchilla81, 2023-07-16:
> @andreschinchilla81 You had right your solution compiles because it is possible to apply `toString()` on every object. Did you find the reason of your issue?

I don't know what the reason is, I tried different versions of SteVe, they all showed the same error. this was the only solution i found

juherr, 2023-07-16:
There is something weird on your screenshot: the stop value is `null`.