steve-community/steve #1191: Websocket IdleTimeout always 45 seconds

tdtuan97, 2023-07-17:
#### Checklist

- [ ] I checked other issues already, but found no answer/solution
- [ ] I checked the documentation and wiki, but found no answer/solution
- [ ] I am running the latest version and the issue still occurs
- [ ] I am sure that this issue is about SteVe (and not about the charging station software or something unrelated to SteVe)

#### Specifications
```
SteVe Version     : 3.6
Operating system  : Docker
JDK               : Docker env
Database          : Mariadb
```

#### Expected Behavior

Websocket keep connection in 2 hours.

#### Actual Behavior

Websocket keep connection in 45s

#### Steps to Reproduce the Problem

  1. Start steve project with env Docker
  2. Using chargeBox (version 1.6J) connect to server using websocket
  3. Waiting for 45 seconds, received message "Transport error
java.nio.channels.ClosedChannelException: null

#### Additional context

...

tdtuan97, 2023-07-17:
[image]

goekay, 2025-12-17:
why the exception happens: https://github.com/steve-community/steve/issues/1913

in short: the client (station) decides to close the connection due to inactivity probably.

we have [a ping-pong mechanism](https://developer.mozilla.org/en-US/docs/Web/API/WebSockets_API/Writing_WebSocket_servers#pings_and_pongs_the_heartbeat_of_websockets) to keep connections alive by sending a ping message in every connection for [every 15 mins](https://github.com/steve-community/steve/blob/master/src/main/java/de/rwth/idsg/steve/config/WebSocketConfiguration.java#L56). the receiver (station) should reply with a pong. i can imagine that this interval is too long for the station, and you could reduce it as you wish.

sorry for the very late answer.